MicrosoftDocs/windows-driver-docs #3435: As of Windows 10,what should the graphics driver do to support miracast

paulpenxu, 2023-03-21:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: ffabf161-7eb2-6a3b-9f1a-f91ea13fedb4
* Version Independent ID: 1c9b5fbf-3be3-85ba-cc94-acc7a9a0101e
* Content: [About wireless displays (Miracast) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/display/wireless-displays--miracast-)
* Content Source: [windows-driver-docs-pr/display/wireless-displays--miracast-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/wireless-displays--miracast-.md)
* Product: **windows-hardware**
* Technology: **display**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2023-03-22:
@paulpenxu, per the Note on this page, drivers are not recommended to implement a custom Miracast stack starting in Windows 10 since the OS ships a built-in stack. As such, the documentation is no longer being updated.

paulpenxu, 2023-03-23:
Thank you for the replay. I am confused aboute what should the driver do to support miracast in Windows 10 since the OS ships a built-in stack in Windows 10.
how does the OS determine whether the graphics driver support miracast or not ?
type " netsh wlan show drivers "in Command Prompt to check if the driver support wireless display.
with  custom Miracast stack , I add "HKR,, MiracastDriverName, %REG_SZ%, Miracast.dll" in registry, it says yes  the graphics driver support wireless display.
how about the built-in stack in Windows 10, should I add the same things to registry?

lorihollasch, 2023-03-24:
I'll reopen this issue and do some digging, @paulpenxu ...

lorihollasch, 2023-03-27:
@paulpenxu, turns out a link to the Win10 information was on this page but buried. I reworked all of the Win8.1 Miracast pages to make the link to Win10+ support details more obvious. Thanks for the issue!

closed 2023-03-27 by lorihollasch

paulpenxu, 2023-03-28:
@lorihollasch , thanks for  the reply. I am tring to make my graphics driver support miracast in win10, I have read the document [Building best-in-class Wireless projection solutions with Windows 10](https://learn.microsoft.com/en-us/windows-hardware/design/device-experiences/wireless-projection), and found that  if my driver 
1) support  WDDM 2.0 or higher 
2) support DHCP 2.0 or higher
3) support H264 and H265 
then, I validate the Wi-Fi and Graphics drivers declare support for Miracast by typing the following at a CMD prompt:
NETSH WLAN SHOW DRIVER
the Graphics Driver will be yes for the wireless display.
like this: 
> Wireless Display Supported: Yes (Graphics Driver: Yes, Wi-Fi Driver: Yes)

Is it right ? I tried but I failed. 
looking forward your help ,thanks in advance.

lorihollasch, 2023-03-28:
@paulpenxu, what exactly is failing? Is your issue beyond your graphics driver? (I assume you didn't add the Win8.1 Registry entry on your Win10 solution?)

I can put out an ask to the engineering team if you can provide specifics about what is failing, but I can't guarantee a response. This repo is for documentation-related issues and work, so I can't offer help beyond that. I encourage you to post your questions in one of the following forums:
* [Microsoft's Windows Q&A](https://learn.microsoft.com/en-us/answers/tags/60/windows)
* [OSR's NTDEV Forum](https://community.osr.com/)

paulpenxu, 2023-03-29:
@lorihollasch , thanks for your reply. 
you are right , I didn't add the the Win8.1 Registry entry on my Win10 solution, and I didn't see this in the document. 
should I add the Registry?

paulpenxu, 2023-03-29:
with custom Miracast stack(wddm 1.3 win8) ,  there is a need to add "HKR,, MiracastDriverName, %REG_SZ%, Miracast.dll" in registry , does win10 need this too ?

lorihollasch, 2023-03-29:
@paulpenxu, don't add it - I was just making sure that it wasn't there! :)