MicrosoftDocs/windows-driver-docs #3438: Unable to register for the Hardware Developer Program

ReeceXW, 2023-03-23:
I'm trying to register for the hardware developer program now that I have an EV Code Signing certificate from GlobalSign. When I visit the starting page:

https://partner.microsoft.com/en-US/dashboard/Registration/Hardware?step=GetStarted

I login with my Azure AD account with global administrator privilege but it takes me straight back to the start page in a loop. I can't progress to fully register for the program.

Because of this I am unable to sign my drivers for release. I am failing to find support anywhere so this is currently my last resort if anybody knows what to do

ReeceXW, 2023-03-27:
> Hi @ReeceXW,
> 
> Sorry to hear you are having trouble with registration. Have you gone to the [support page](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-support#technical-support-issues) and followed the directions indicated?
> 
> 
> 

Hi, thanks for replying, I've tried about everything. I haven't gotten a reply from support or found any useful information online.

jmzjmzjmz, 2023-07-07:
Having the same loopback issue... @ReeceXW have you had any luck?

anaharris-ms, 2023-07-10:
Hi @jmzjmzjmz,

I will look into this again to see if I can get any more information.

anaharris-ms, 2023-07-11:
Hi @jmzjmzjmz 

The support team was not able to find any support tickets for the email address that you provided. Can you follow the guidance below to open a new ticket? Then, please share the ticket number with me here, so that I can expedite.  

Go to [https://developer.microsoft.com/en-us/windows/support/?tabs=Contact-us](https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdeveloper.microsoft.com%2Fen-us%2Fwindows%2Fsupport%2F%3Ftabs%3DContact-us&data=05%7C01%7Canaharris%40microsoft.com%7Cd12f55ac8a5f41b414f908db82262819%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C638246872633812272%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=IkKQPkJi638RfadyG76VXUQ50JMEk1L3CmKekLLI1To%3D&reserved=0)
Under Non-Technical Support – Dashboard heading, click [Submit an incident](https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fsupport.microsoft.com%2Fsupportrequestform%2Fe7a381be-9c9a-fafb-ed76-262bc93fd9e4&data=05%7C01%7Canaharris%40microsoft.com%7Cd12f55ac8a5f41b414f908db82262819%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C638246872633812272%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=vm8wEFL3lE%2F88uRVm0fd791FnUN%2BL4xaXRpZO7iWEbc%3D&reserved=0)
Thank you!

jmzjmzjmz, 2023-07-14:
Hi @anaharris-ms @ReeceXW , we discovered the issue! 

The redirect loop happens when attempting to use a standard Microsoft Azure account (ending in our domain name). The fix was to log in via our dylan@accountnamehere.OnMicrosoft.com account. 

It would be helpful to throw some kind of warning/error that the user doesn't have an azure active directory account.

In any case,  Thanks for your quick response

closed 2023-10-24 by anaharris-ms