MicrosoftDocs/windows-driver-docs #3442: Win11 PDB file of ntoskrnl.exe etc missing

BlackINT3, 2023-03-26:
SYMSRV:  HTTPGET: /download/symbols/ntkrnlmp.pdb/E9BEBA2FE31582704229E51F816084331/ntkrnlmp.pdb
SYMSRV:  HttpQueryInfo(HTTP_QUERY_CONTENT_LENGTH): 800C2F76 - ERROR_HTTP_HEADER_NOT_FOUND
SYMSRV:  HttpQueryInfo: 80190194 - HTTP_STATUS_NOT_FOUND

`
OS: Win11 22H2
10.0.22621.1483 (WinBuild.160101.0800)
File Version 10.0.22621.1483

Name ntkrnlmp.exe, as well as win32kfull.sys etc.
`

tedhudek, 2023-03-27:
Hi @BlackINT3 thanks for reaching out.  Is your feedback related to a documentation page?  If so can you please include the link of the page?

tedhudek, 2023-03-27:
Thanks @BlackINT3, we are now tracking this issue internally.  We appreciate the report.

closed 2023-03-27 by tedhudek

tedhudek, 2023-03-28:
@BlackINT3 -- I think the symbols should be available now.  Can you please try again and let me know if it works?  Thanks again for helping us work this through.

BlackINT3, 2023-03-29:
@tedhudek Yeah, it works. nice job!