MicrosoftDocs/windows-driver-docs #3445: Bugcheck 0x1C Reference Help information is not helpful

rarndtjr, 2023-03-28:
[The help documentation reference info 'This bug check appears very infrequently.' for Bug Check 0x1C PFN_REFERENCE_COUNT is not helpful in the least. 

At minimum the bugcheck help / reference  documentation needs to include the following:

1. An explanation about what exactly the bugcheck 1C is.
2. What scenarios can cause the failure to occur.
(Bad RAM, memory corruption, storage, IO paging, cache coherency issues, drivers trashing memory address regions, processor issues, etc.)
3. Possible methods to debug the issue. 

Note that the same unhelpful message appears at the URL below and also within WinDBG help reference.  If you need a memory dump to help investigate to generate better documentation then we might be able to help with this through our MS TAM / OPM.

Thanks!]

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: e81ec1c7-20c2-62b6-07b3-1dc8e9b7a552
* Version Independent ID: cf863e65-92e5-cfd1-6be2-a6d9ac8d75c2
* Content: [Bug Check 0x1C PFN_REFERENCE_COUNT - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1c--pfn-reference-count)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1c--pfn-reference-count.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1c--pfn-reference-count.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2023-04-07:
@rarndtjr - Thanks for this issue, I expanded the description of this bug check. 

In my experience, some of the dump files for this bug check that point to the networking stack. Using !analyze to see if a specific bit of code is implemented, such as network driver, or a downstream network protocol. One challenge with memory errors is that it may have been previous code that led to the corruption, for example it didn't mark an area of memory as freed. So other code later hits that area of memory, and fails, so finding the root cause can be a challenge. Using multiple dump files, or if possible, having a debugger attached, when it fails, can increase the odds of tracking down the culprit. Normal suspects would be all network related code, such as virus filters or backup software, that scan traffic. 

General information on attacking bug checks is not repeated in each topic, but rather in topics such as: [Blue screen data](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/blue-screen-data)
Also the defrag tools show walks through a number of tactics: https://learn.microsoft.com/en-us/shows/Defrag-Tools/

One technique of note would be to enable driver verifier, to have it attempt to locate bad driver behavior, that later leads to memory corruption. 

Good luck with determining the cause, and thanks again for filing this docs issue.

closed 2023-04-07 by DOMARS

rarndtjr, 2023-04-11:
Hi Don,

Thanks for the response.  I am familiar with the KD / WinDbg and kernel debug methods and if I need additional assistance I can use our OEM support agreement. My concern is that there is virtually no meaningful details or information related to bugcheck C1 other than the bugcheck is "very rare" which is effectively of little value.

I was hoping to bring this to the attention of the WinDbg team to provide more meaningful detail and to update the debug documentation within WinDbg and online via feedback vs opening an support case / bug.

Is this something you can help address?

Thanks.

Ron Arndt