MicrosoftDocs/windows-driver-docs #3456: "Porting NDIS miniport drivers to NetAdapterCx" doesn't talk about setting the supported packet filter types

guyharris, 2023-04-05:
[The "Porting NDIS miniport drivers to NetAdapterCx" page](https://learn.microsoft.com/en-us/windows-hardware/drivers/netcx/porting-ndis-miniport-drivers-to-netadaptercx) doesn't directly talk about how to set the types of packet filters that 802.3 (or 802.5, but that's dead) (and what about 802.11?) drivers support.

[The "Device initialization" section](https://learn.microsoft.com/en-us/windows-hardware/drivers/netcx/porting-ndis-miniport-drivers-to-netadaptercx#device-initialization) appears to be the most appropriate section, but the page it suggests, ["Device and adapter initialization"](https://learn.microsoft.com/en-us/windows-hardware/drivers/netcx/device-and-adapter-initialization), doesn't mention [`NetAdapterSetReceiveFilterCapabilities()`](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/netadapter/nf-netadapter-netadaptersetreceivefiltercapabilities).

Other routines useful in initializing the driver may also not be mentioned there.

I note this because nmap/npcap#628 may be more likely to show up on adapters with NetAdapterCx drivers, and the Npcap driver, when asked to set an adapter's packet filter, turns off all bits in the filter that aren't in what the Npcap driver believes to be the SupportedPacketFilters, and, if that value isn't correct, it may turn off so many bits that the resulting filter is considered invalid.  If the adapter driver doesn't call `NetAdapterSetReceiveFilterCapabilities()`, perhaps because the driver writer didn't see it mentioned, the value will be whatever default NetAdapterCx provides.

If we look at the fields in [the `NDIS_MINIPORT_ADAPTER_GENERAL_ATTRIBUTES` structure](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ndis/ns-ndis-_ndis_miniport_adapter_general_attributes), the equivalent NetAdapterCx routines for some of them are:

- `MtuSize`: [`NetAdapterSetLinkLayerMtuSize()`](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/netadapter/nf-netadapter-netadaptersetlinklayermtusize)
- `MaxXmitLinkSpeed`, `MaxRcvLinkSpeed`: [`NetAdapterSetLinkLayerCapabilities()`](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/netadapter/nf-netadapter-netadaptersetlinklayercapabilities)
- `XmitLinkSpeed`, `RcvLinkSpeed`, `MediaConnectState`, `MediaDuplexState`, `SupportedPauseFunctions`, `AutoNegotiationFlags`: [`NetAdapterSetLinkState()`](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/netadapter/nf-netadapter-netadaptersetlinkstate)
- `SupportedPacketFilters`, `MaxMulticastListSize`: [`NetAdapterSetReceiveFilterCapabilities()`](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/netadapter/nf-netadapter-netadaptersetreceivefiltercapabilities)
- `MacAddressLength`, `PermanentMacAddress`: [`NetAdapterSetPermanentLinkLayerAddress()`](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/netadapter/nf-netadapter-netadaptersetpermanentlinklayeraddress)
- `MacAddressLength`, `CurrentMacAddress`: [`NetAdapterSetCurrentLinkLayerAddress()`](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/netadapter/nf-netadapter-netadaptersetcurrentlinklayeraddress)

so all of those should probably be mentioned.  There may be other than I've missed.
  
---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: bd8d1135-8e43-d2bd-4398-4b04cd727ea1
* Version Independent ID: 24d46a09-1175-b223-be59-a7317e925038
* Content: [Porting NDIS miniport drivers to NetAdapterCx - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/netcx/porting-ndis-miniport-drivers-to-netadaptercx)
* Content Source: [windows-driver-docs-pr/netcx/porting-ndis-miniport-drivers-to-netadaptercx.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/netcx/porting-ndis-miniport-drivers-to-netadaptercx.md)
* Product: **windows-hardware**
* Technology: **netcx**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

jtippet, 2023-04-10:
I agree that this could be demonstrated explicitly in the docs - [this page](https://learn.microsoft.com/en-us/windows-hardware/drivers/netcx/device-and-adapter-initialization) should demonstrate `NetAdapterSetReceiveFilterCapabilities`. Our sample driver already does do this, but the docs should reflect that.

(In general, where NDIS6 used a field to set the initial value during NIC initialization, plus an OID to query/change it at runtime, NetAdapter has only a single function that can set the initial or runtime-changed value.)

aviviano, 2023-05-24:
Thanks @guyharris for opening this issue. We've opened an internal work item to improve the device and adapter initialization page. I'll close this issue now and let you know once the updates are published.
Thanks for helping us improve our docs.

closed 2023-05-24 by aviviano