MicrosoftDocs/windows-driver-docs #3477: Clarify parameter 4 value

jfbradfo, 2023-05-02:
The documentation for the 0x3b bug check has an internal conflict with respect to Parameter 4.

In the table of parameters, parameter for is defined as "0 (not used)".   However, just below that in the common exception code list, for 0xc0000005, the document states (parenthetically) "Parameter 4 of the bug check is the address that the driver attempted to access."

Which is correct?

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: b19eb27e-40b0-c765-d826-8b9962e4c1e6
* Version Independent ID: f9231818-2d1d-be08-7ce2-92b97edd5613
* Content: [Bug Check 0x3B SYSTEM_SERVICE_EXCEPTION - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x3b--system-service-exception)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x3b--system-service-exception.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x3b--system-service-exception.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2023-05-18:
@jfbradfo - Thanks for seeing this inconsistency and filing this issue. It is not the case that the param 4 would contain this address. Parm 4 is not used. So, I updated the doc to remove this contradictory information. Thanks for having a sharp eye, and helping to improve the bug check documentation.

closed 2023-05-18 by DOMARS

jfbradfo, 2023-05-25:
Just validated the production docs.  Thanks for updating so quickly!