MicrosoftDocs/windows-driver-docs #3482: PROCESSOR_START_TIMEOUT is not here

TachUDF, 2023-05-05:
PROCESSOR_START_TIMEOUT is not here

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: ddc632b2-f4e1-b784-dd1e-878feac077f9
* Version Independent ID: 7a6866f4-3333-f2d1-86d7-dbb3d00925e0
* Content: [Bug Check Code Reference - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-code-reference2)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-code-reference2.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-code-reference2.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2023-05-26:
@TachUDF - Thanks for this issue describing the need to document the PROCESSOR_START_TIMEOUT (1df) bug check code. I have started to work to add this to the bug check code documentation and will be tracking this work item internally.

It looks like the docs page will just mirror what !analyze displays, which is that "a processor failed to start in the allowed time". So until the docs are completed, I would recommend opening the dump file in WinDbg and using the [!Analyze](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/using-the--analyze-extension) command to display the parameters for your bug check.  This process is described in [Open a dump file with WinDbg](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/opening-a-crash-dump-file-using-windbg).

Good luck with your bug check code troubleshooting and thanks for passing on that you are running into PROCESSOR_START_TIMEOUT.

closed 2023-05-26 by DOMARS

DOMARS, 2023-06-16:
@TachUDF - Wanted to let you know that the docs page for PROCESSOR_START_TIMEOUT was published today.

https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1df--processor-start-timeout

Thanks for providing input on the bug check codes, and your interest in improving the docs.