MicrosoftDocs/windows-driver-docs #3483: NET_REQUEST_QUEUE_CONFIG related functions can not be searched on the microsoft website

optlee, 2023-05-06:
Net Request Queue related functions can not be searched on the microsoft website. Also can not be searched on the Google.
are they(functions & macros) discarded or not support yet?
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 2616c6f5-e6fd-fbf3-8e1c-c3c11cb18558
* Version Independent ID: f75eb509-1e6d-029b-ee0c-d7525d2beed9
* Content: [Building a NetAdapterCx client driver - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/netcx/building-a-netadaptercx-client-driver)
* Content Source: [windows-driver-docs-pr/netcx/building-a-netadaptercx-client-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/netcx/building-a-netadaptercx-client-driver.md)
* Product: **windows-hardware**
* Technology: **netcx**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

optlee, 2023-05-10:
no any feedback?

aviviano, 2023-05-10:
Hi @optlee,
The NET_REQUEST_QUEUE_CONFIG functions were part of the preview version of NetAdapterCx and are not supported in 2.0+.
What are you trying to accomplish with that API?

optlee, 2023-05-11:
I want to port a ndis miniport wifi driver to NetadapterCx model.  But there is no api for OID command, which is registered to upper level. So does NetadapterCx support wireless netcard which OID is needed?

aviviano, 2023-05-11:
To write a wifi driver you need to use WiFiCx in conjunction with NetAdapterCx. Have you reviewed the WiFiCx documentation?  [Writing a WiFiCx client driver](https://learn.microsoft.com/en-us/windows-hardware/drivers/netcx/writing-a-wificx-client-driver)

optlee, 2023-05-12:
ok, I have read WifiCx docs. But WifiCx only support the OS from Windows11. Then Which model should be used on Win10?

aviviano, 2023-05-15:
You'll need to use WDI to load the driver on Windows 10: https://learn.microsoft.com/en-us/windows-hardware/drivers/network/wdi-miniport-driver-design-guide 
As you mention, you can only load a WiFiCx driver on Windows 11. 
Thanks @optlee for opening this issue. Please open a new issue if you have further questions.

closed 2023-05-15 by aviviano