MicrosoftDocs/windows-driver-docs #3494: Can Not Uninstalled Windows Driver Kit For Windows 8.1 Ver.8.100.26846

bipsmartec, 2023-05-19:
[Enter feedback here]
Can Not Uninstalled Windows Driver Kit For Windows 8.1 Ver.8.100.26846.
Message Windows's Contents is "Setup cannot locate SDK ARM Additions EULA-x86_en-us.msi"
We don't own this version's setup media.

So we want to re-install and after then  re-uninstall. But now ,this version's media cannot get from your web site.
How can we uninstall Windows Driver Kit For Windows 8.1 Ver.8.100.26846?
 

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 3f5d6d55-cb35-6b9e-f8ca-e75f70826dc3
* Version Independent ID: 1d913516-b1e0-e872-c337-e30ed5dffce8
* Content: [Previous WDK versions and other downloads - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads)
* Content Source: [windows-driver-docs-pr/other-wdk-downloads.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/other-wdk-downloads.md)
* Product: **windows-hardware**
* Technology: **whats-new-and-downloads**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2023-05-20:
Thanks for the feedback, @bipsmartec and sorry you are having trouble.  Can you clarify, is the problem that you can't uninstall WDK 8.1, or that you can't install WDK 8.1?  (or both?)

bipsmartec, 2023-05-20:
Thank you for your reply.

 

I can install WDK8.1.

However, I am unable to uninstall WDK8.1.

 

The version of WDK8.1 installed on my PC is 8.100.26846 and the uninstall process seems to require this version of media.

 

If I download WDK8.1 uploaded on your website and specify it as the msi requested during the uninstall process, the uninstall process will not proceed.

I think it is because the versions are different.

 

If I could get the WDK8.1 media for version 8.100.26846, I would be able to finish the uninstall process without stopping.

 

Best Regards.

Isao Doumoto

tedhudek, 2023-05-22:
Hi @bipsmartec sorry but I don't have a lot to offer you.  Someone did raise #3480 recently which has a link to a Stack Overflow article that has links to 8.1 kits that are online.  I am not sure if any of these are the version you are looking for.  You might be better off manually deleting things or re-imaging the machine in question, if that's possible.  I'm going to go ahead and close out this issue but feel free to reply here if you have any follow up questions.

closed 2023-05-22 by tedhudek