MicrosoftDocs/windows-driver-docs #3501: VS2022 (17.6.1) shows Driver Testing Error Dialog on startup after installing WDK 10.1.22621.382.

carl-woodward, 2023-05-23:
Environment:
* Windows 11 AMD64, Build 22621.1702.
* VS 2022 Community 17.6
* SDK: 10.1.22621.755
* WDK: 10.1.22621.382.

Issue:
After installing the Windows SDK and WDK above, VS 2022 Community 17.6.1 shows an error dialog on startup:

"Driver Testing
Could not load test information. Error: Object reference not set to an instance of an object".

[image]

How do I get to the bottom of why this is happening? I have little confidence that the WDK is properly installed, and this puts the correctness of my build environment into doubt.

Notes:
* I have repaired and reinstalled VS, WDK, SDK.
* I have a number of installations of these versions and they work correctly.

tedhudek, 2023-06-05:
Thanks for the report, Carl and apologies for the delay.

The behavior you describe does match a bug that we have seen previously  Can you confirm if you are able to dismiss the warning and successfully build a driver, or does it actually prevent you from building?

tedhudek, 2023-06-12:
Closing out for now, please re-open if needed

closed 2023-06-12 by tedhudek