MicrosoftDocs/windows-driver-docs #3502: "write a hello world windows driver" Inf2Cat Unknown Failure

lewisrusselljohn, 2023-05-24:
Inf2Cat Tool Output:
........................
Signability test complete.

Errors:
None

Warnings:
None

An attempt was made to load a program with an incorrect format. (Exception from HRESULT: 0x8007000B)
Signability test failed.

Can anybody help me?

lewisrusselljohn, 2023-05-24:
HI, an update:
After a reboot this morning, the Inf2Cat error has changed to the following:

Inf2Cat Tool Output:
........................
Signability test failed.

Errors:
22.9.7: DriverVer set to a date in the future (postdated DriverVer not allowed) in KMDFHelloWorld\kmdfhelloworld.inf.

Warnings:
None

I've tried setting "User localtime" in vs2019, like the tutorial suggested but I get the same error. I'll try a reboot again.

lewisrusselljohn, 2023-05-24:
Changed the local time of the system to be Eastern US. No longer getting the DriverVer error but it's now gone back to the first issue.

lewisrusselljohn, 2023-05-25:
Ok, well, I could not find a real solution for the problem but I uninstalled VS2019 and the WDK I was using and installed the latest WDK and Windows 11 SDK and now it all builds right.

I think maybe I missed WDK warnings about compatibility the first time I installed whichever WDK version I used first.

closed 2023-05-25 by lewisrusselljohn

tedhudek, 2023-05-31:
Hi @lewisrusselljohn, thanks for reaching out and apologies for taking a while to get back to you.  Glad you got the issue fixed.  I was able to confirm that you ran into a known bug.  I have added two possible workarounds to a new "Troubleshooting" section on https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/inf2cat.  That update should appear on the live site within a few minutes.  So if you ever run into it again, those are steps you can try.  Thanks again for caring about the docs!

lewisrusselljohn, 2023-06-01:
Ah, that's cool, thanks for following that up @tedhudek ! It's good to know there's a few workarounds for it.