MicrosoftDocs/windows-driver-docs #3506: The ISO does not install anything to c: so are we supposed to run inside the E: drive letter every time?

markyoung2u, 2023-05-26:
I don't understand. Mount the EWDK_ni_release_svc_prod1_22621_220804-1759.iso, run SetupVSEnv and then invoke devenv.exe from there. Are we supposed to do this every time? Is there not a way to install to the hard drive?

HollisTech, 2023-05-26:
It doesn't install anything, it is a standalone build environment. Yes you need to run the setup script(s) in each shell that is going to use the kit. It is ideal for build machines, but I also use it as a standard dev environment within vscode.

markyoung2u, 2023-05-26:
I've been fighting VS 2022 install issues. I installed it about a year ago. Now it's all changed. I can see how MSFT might love a mounted ISO solution - read-only, never corrupted, but I would hope that MSFT would also provide a script to use to invoke VS 2022 with the ISO. I'm looking at making it mount at a fixed drive letter so all my peers can do the same thing. E: may move around, etc. To answer your question, yes, please say something - maybe HollisTech or someone could provide a sample script to use as a VS 2022 shortcut - but the way it stands, it's a bit unclear

HollisTech, 2023-05-26:
@markyoung2u  the setupVSEnv.cmd script in BuildTools is intended to launch a shell from which you can start VisualStudio with the proper environment set.. You first run SetupBuildEnv.cmd, then SetupVSEnv.cmd. However be warned that each EWDK release is tied to a specific version of VS, so using it this way puts you back to having a specific VS version dependency. This is why I use it with VSCode instead. That way I can have any version of VS installed on my build system and still have a complete control of which version of the WDK I am using. Best of both worlds as far as I am concerned.  As far as detecting the EWDK volume path, that is as trivial as searching for a path <a-z>:\EWDK\BuildEnv, i.e.
``` powershell

Get-PSDrive -PSProvider 'FileSystem' | ForEach-Object {
        $drv = "$($_.Name):"
        if (Test-Path -Path "$drv\EWDK\BuildEnv" -PathType Container) {
             $script:defaultProfessionalToolsPath = "$drv\EWDK\BuildEnv"
              Write-Host "EWDK found at $drv\EWDK"
              break
         }
 }

```

markyoung2u, 2023-05-28:
Maybe this not the right forum but a big part of this is that I cannot seem to get the entire ISO downloaded. I was not even aware there was any issue as EDGE gave me no indication the 1st time that I did not get entire 15.9 GB image when I only got 9GB 
9,287,778,304 EWDK_ni_release_svc_prod1_22621_220804-1759.iso
I moved/saved that iso and tried again and got an image just less than 9GB (again with EDGE). I tried 3rd time with Chrome and got 9,998,090,239 EWDK_ni_release_svc_prod3_22621_230511-2102.iso (newer 17.1.6 version) but neither 2nd nor 3rd version can be mounted, File Explorer says they are corrupt why I try to mount, but neither Edge nor Chrome gave me any download error message? Is there some way of downloading more reliably on a home 6 MBs network than using a browser? Can I somehow use robocopy?

rodwiddowson, 2023-05-29:
Welcome to the world where you are disadvantaged if you do not have gigabps internet and software is compiled, released and ‘tested in production’…

  I have 3mbs (per channel)and for a long time the only successful way I have been able to download ISO’s and memory dumps is to RDC to a well connected machine (aws will do it) download there and then zip into a multiparty (75kb per part) file.  Then use onedrive to transfer that locally - onedrive is better at picking up from where it left off and if you lose 50kb it’s not the end of the world.  It also understand how to drive multiple channels (so I can max out both my 3mbs ADSL links)

Note that edge and chrome are the same thing these days. You might get more robust results with cUrl but it sounds like you link is bouncing.  Firefox is another idea

good luck

markyoung2u, 2023-05-30:
I came into work and downloaded for 4th time. Even here it got a Network Error mid-download. I saw it and clicked a "|>" button to Resume (never saw this before, wondering if it just goes away by itself?). It appeared to pick up where it left off and this time it completed, and I got the full ISO: 17,045,657,600 EWDK_ni_release_svc_prod3_22621_230511-2102.iso. I was able to mount it and invoked devenv.exe from the D: drive that the ISO mounted to, but the download page says it's supposed to be 17.1.6 but it reports 17.1.5 ?
[image]

tedhudek, 2023-06-01:
So sorry this has been such an ordeal @markyoung2u.  

If you have any suggestions or recommendations specific to the docs, anything we could include that might help the next person to come down this path, please do me know.  Happy to leave this issue open a bit since it seems it's been useful to interact.  I wonder if you might get more results by trying a wider forum, for example:

* [windows-hardware-wdk](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk.html)
* [OSR's NTDEV forum](https://community.osr.com/categories/ntdev)

markyoung2u, 2023-06-02:
1. MSFT should clarify this is not an installation step at all, it is something you must do every time you want to work on a driver with VS 2022. I.e. downloading the ISO is the install part, the rest is the execute-every-time part.
2. Even on a high-speed connection the almost 16GB download gets and error (but can be resumed successfully) and on slower connections it simply does not work. I'm sure can understand that a warning should be added to save thousands of man hours chasing down issues if it actually mounts but is not the full image, or retrying the download if mounting reports a corrupted image. This is especially bad since so many people work from home now. Please provide a direct link so something like robocopy can download, verify, and restart where it left off on errors, etc.
3. Fix the secondary step https://learn.microsoft.com/en-us/legal/windows/hardware/enterprise-wdk-license-2022 to change "17.1.6 (Recommended)" to "17.1.5 (Recommended)" or fix the ISO itself to actually report 17.1.6
Ted, I appreciate your wanting to get this resolved, as do I. I'm still not 100% successful after weeks of working on this and the end is still not in sight...

tedhudek, 2023-06-06:
Hi Mark, I've just pushed an update to https://learn.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk#enterprise-wdk-ewdk that clarifies which is the part to execute every time.  

I did download the one labelled 17.1.6 myself and was able to reproduce the behavior you saw, i.e. after I downloaded it and mounted it, the command prompt that came up said 17.1.5.  I'm reaching out internally to see what we can do about that.

markyoung2u, 2023-06-06:
I could not see any change you made?

- I noticed the link https://learn.microsoft.com/en-us/visualstudio/releases/2022/system-requirements has 17.5 as the latest, not 17.6 although 17.6 and  future 17.7 are mentioned later

[Visual Studio 2022 version 17.5 Release Notes](https://learn.microsoft.com/en-us/visualstudio/releases/2022/release-notes-v17.5)
- I've always thought "Enterprise WDK (EWDK)" was not labeled "Step 4" because it is optional, not needed if you do not develop drivers, but maybe it could be added. Just an idea, not asking for it.

-  I looked at links and was trying to figure out the .NET Framework version. Below says to look at registry "HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\NET Framework Setup\NDP\v4\Full" and next para talks about the "Release" REG_DWORD value, but should be Client\Release. My value 533325 indicates .NET Framework 4.8.1
https://learn.microsoft.com/en-us/dotnet/framework/migration-guide/how-to-determine-which-versions-are-installed

[image]

markyoung2u, 2023-06-06:
HollisTech - Have you actually tried win11 with VS 2020 mounted the ISO and compiled a driver?

tedhudek, 2023-06-07:
Hi @markyoung2u, 

On https://learn.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk#enterprise-wdk-ewdk, I added the sentence that starts "Once you have downloaded the ISO" to indicate that the download is a one-time operation and is then followed by the steps to repeat each time you want to build.  I also added step 5 about ejecting the ISO to make it clear that these steps are repeatable.

You're right that EWDK is not labelled step 4 because it's an alternate way to use the kit.  I added a sentence just below the heading to indicate that it is an alternative.

Thanks again for letting us know, we've updated https://learn.microsoft.com/en-us/legal/windows/hardware/enterprise-wdk-license-2022 to indicate that the first link is indeed 17.1.5.

Finally, https://learn.microsoft.com/en-us/dotnet/framework/migration-guide/how-to-determine-which-versions-are-installed is owned by a different documentation team (and is sourced out of a different repo).  If you would like to alert them to the issue you found there, you can select "Submit and view feedback for This page" at the bottom of that page.

On behalf of the driver doc team, thanks again for reaching out and best of luck sorting through these issues.  Again, the forums I linked to above can be quite helpful for this type of query.

closed 2023-06-07 by tedhudek

markyoung2u, 2023-06-26:
I have been under impression that EWDK is needed to build drivers, now you are telling me it is an alternate way. I will pursue this elsewhere, you can close this issue

Mark

Thanks again for letting us know, we've updated https://learn.microsoft.com/en-us/legal/windows/hardware/enterprise-wdk-license-2022 [learn.microsoft.com]<https://urldefense.com/v3/__https:/learn.microsoft.com/en-us/legal/windows/hardware/enterprise-wdk-license-2022__;!!LpKI!j5hdKA9Xeui27zDzowab0CzKdsKyNczDMXnOY38e3PzeS8DwqnsTa1qbpDsXBVsNBvToexZ4Xce9uHuGn-S8o4rPsGY$> to indicate that the first link is indeed 17.1.5.

Finally, https://learn.microsoft.com/en-us/dotnet/framework/migration-guide/how-to-determine-which-versions-are-installed [learn.microsoft.com]<https://urldefense.com/v3/__https:/learn.microsoft.com/en-us/dotnet/framework/migration-guide/how-to-determine-which-versions-are-installed__;!!LpKI!j5hdKA9Xeui27zDzowab0CzKdsKyNczDMXnOY38e3PzeS8DwqnsTa1qbpDsXBVsNBvToexZ4Xce9uHuGn-S8LzVFQDE$> is owned by a different documentation team (and is sourced out of a different repo). If you would like to alert them to the issue you found there, you can select "Submit and view feedback for This page" at the bottom of that page.

Internal Use - Confidential