MicrosoftDocs/windows-driver-docs #3512: Not find the webpage  "Submit new hardware"

yinming-shenzhen, 2023-06-01:
I could not find the webpage  "Submit new hardware", Could you give the link? thanks

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: a9288fd9-bb24-631a-b6f8-6b2f5313fd17
* Version Independent ID: a9288fd9-bb24-631a-b6f8-6b2f5313fd17
* Content: [Create a new hardware submission - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-create)
* Content Source: [windows-driver-docs-pr/dashboard/hardware-submission-create.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/hardware-submission-create.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @anaharris-ms
* Microsoft Alias: **anaharris**

anaharris-ms, 2023-06-01:
Hi @yinming-shenzhen,

Sorry you are having trouble. Hopefully, the following screenshot helps!

[image]

yinming-shenzhen, 2023-06-02:
Hi, anaharris-ms, Thanks for your comment.

I have met a problem when do windows driver certificate. 

[image]

The EV certificate is purchased by our company by the legal way and I am sure the EV is owned only by us. But Why the website tell me the EV  certifate has already been registered? Could you help me to check? Thank you very much.

anaharris-ms, 2023-10-24:
Hi @yinming-shenzhen ,

Please contact customer support [here](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-support#technical-support-issues) for cert submission issues.

closed 2023-10-24 by anaharris-ms