MicrosoftDocs/windows-driver-docs #3538: USB Device Layout Docs clarification

Veverke, 2023-06-05:
In the 2nd paragraph, the sentence I singled out in bold:

A USB configuration defines the capabilities and features of a device, mainly its power capabilities and interfaces. The device can have multiple configurations, but only one is active at a time. The active configuration isn't chosen by the USB driver stack, but might be initiated by an application, a driver, the device driver. **The device driver selects an active configuration**.

1. The active configuration is not set by the USB driver stack. Who sets it then ? Is this not-determinable ?
2. The docs mentions  "an app, driver, device driver" - as a POSSIBLE source that sets what the active configuration is ? Or the source will always be one of these 3 ? 
3. The last sentence says the device driver selects an active configuration. 
3.1 Does this mean that ultimately it is the device driver that selects the active configuration? What does select an active configuration" mean ? 
3.2 Something else set such configuration as active ? If not, it should read "device driver **_sets the_** active configuration". If it does not set, what is there to be selected if the active configuration is already set and there can always be only one active configuration ?

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: de40fe23-93bf-f9a0-5a77-6a1d25850142
* Version Independent ID: a0543280-5e1f-4236-c9b7-896835369cb8
* Content: [USB device layout - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-device-layout)
* Content Source: [windows-driver-docs-pr/usbcon/usb-device-layout.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/usb-device-layout.md)
* Product: **windows-hardware**
* Technology: **usbcon**
* GitHub Login: @mhopkins-msft
* Microsoft Alias: **mhopkins**

Veverke, 2023-06-05:
I did not notice that part of my question "What does select an active configuration mean" is addressed afterwards. There is a typo there, though (redundant "or - marked in bold).

So ultimately I understand that
1. The device driver IS the one and only one responsible for selecting an active configuration
2. Selecting an active configuration is the same as setting the active configuration.

> So what does it mean to select an active configuration? During device initialization, the device driver for USB device must select a configuration, one or more **or** interfaces within that configuration, and an alternate setting for each interface. Most USB devices don't provide multiple interfaces or multiple alternate settings. For example, the OSR USB FX2 Learning Kit device has one interface with one alternate setting and three endpoints. For more information about the learning kit, see [OSR Online](https://www.osronline.com/).

mhopkins-msft, 2023-06-09:
Hi @Veverke 

Thank you for reaching out to the driver documentation team. I've fixed the typo that you pointed out. Thank you for that feedback.

The men and women to staff the driver documentation team are documentation professionals with limited resources and are unable to respond to individual requests for support.

You might try posting your question on the [OSR Developer Community forums](https://community.osr.com/). There are many other Windows system developers there who may be able to assist you.

You could also post your question on the [Microsoft Q&A site](https://learn.microsoft.com/en-us/answers/).

I wish you the best in resolving your questions. If you have a recommendation for updating the documentation, please feel free to reopen this issue.

Thank you,
 ~ *Mark*

closed 2023-06-09 by mhopkins-msft