MicrosoftDocs/windows-driver-docs #3553: TAEF Executing Tests

Elevoc-Shaoting, 2023-06-09:
TAEF needs to compile the code into a dll file when executing Te test C++ code, but I don’t see the project properties of the compiled dll, I don’t know this step, please help me explain, thank you very much!

Elevoc-Shaoting, 2023-06-12:
I have some compilation errors when compiling the cpp files in the Windows-driver-samples-master\audio\tests\HLK\ThirdPartyApoTests directory and I can't find the reason: such as IEndpointUtility, IMMEndpointInternal.
Is there any other header file I should include? But it's not clear to me where these header files are eg: #include <nt.h>
#include <ntrtl.h>
#include <nturtl.h>

DOMARS, 2023-06-16:
@Elevoc-Shaoting - Thank you for reaching out to the driver documentation team and passing on your experiences with compiling the HLK ThirdPartyApoTests. 

The men and women to staff the driver documentation team are documentation professionals with limited resources and are unable to respond to individual requests.

Typically, the HLK tests are used by vendors that are working to certify drivers for commercial use. I would probably look at the sample code in the topics here for C++ and C#, and start with those smaller samples.

https://learn.microsoft.com/en-us/windows-hardware/drivers/taef/authoring-tests

Once that is building, you could move code you are interested in from the HLK tests to the smaller samples.

If you have other questions, you might try posting your question on the [OSR Developer Community forums](https://community.osr.com/). There are many other Windows system developers there who may be able to assist you.
You could also post your question on the [Microsoft Q&A site](https://learn.microsoft.com/en-us/answers/tags/60/windows).
I wish you the best in resolving your questions.

closed 2023-06-16 by DOMARS