MicrosoftDocs/windows-driver-docs #3556: Ethernet encapsulation required for LSO/RSC?

jwhited, 2023-06-13:
> Miniport adapters must support Ethernet encapsulation for all of the types of task offload that they support. The other types of encapsulation are optional.

This snippet suggests that the driver must also be functioning at an Ethernet layer, to make use of any task offloads, even those for transport protocols. Is this the case?

Other docs hint at this, e.g. NDIS_OFFLOAD struct docs. Can a miniport driver/adapter operating with a media type of `NdisMediumIP` leverage transport layer task offloads such as LSO/RSC?

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: f26b5e10-4d40-1ebe-6b49-4548149d08d3
* Version Independent ID: f6010cdb-5f01-8c40-b47f-b6fc055b4f61
* Content: [OID_TCP_OFFLOAD_CURRENT_CONFIG - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/network/oid-tcp-offload-current-config)
* Content Source: [windows-driver-docs-pr/network/oid-tcp-offload-current-config.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/oid-tcp-offload-current-config.md)
* Product: **windows-hardware**
* Technology: **network**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2023-06-26:
Hi @jwhited,
Thanks for opening this issue. Here is the response from the development team:
The documentation is correct – miniport adapters must support ethernet encapsulation to support task offload. 

TCPIP expects the encapsulation type to be either NDIS_ENCAPSULATION_IEEE_802_3 or NDIS_ENCAPSULATION_IEEE_LLC_SNAP_ROUTED for offloads to be enabled. I don’t believe either of these apply for raw IP packets.

Do you have any further questions?

jwhited, 2023-06-26:
> Hi @jwhited, Thanks for opening this issue. Here is the response from the development team: The documentation is correct – miniport adapters must support ethernet encapsulation to support task [...]

No, thank you.

closed 2023-06-26 by jwhited