MicrosoftDocs/windows-driver-docs #3558: Update docs on NetMon to be more Specific

stellarpower, 2023-06-16:
Please could you review the details in:
- [Using the Network Monitor Tool](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/portable/using-the-netmon-tool.md)
- [How to install Netmon and USB ETW Parsers](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/how-to-install-netmon-and-the-netmon-usb-parser.md)
? 

After several hours and gigabytes downloaded I have come to the conclusion (still waiting for it to extract) that the WDK version required for the required files to be present is at least 22H2, but the former doc isn't well-formatted in places to start with (code not highlighted in the ordered list, etc.), and referred to a path with an apparently random version of the kit as installed on the user's system. If a specific version of the WDK, and if the full process of installing Visual Studio, then the SDK, then the WDK as mentioned on the [rather lengthy page](https://learn.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk) and [long compatibility matrix](https://learn.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads) that links to, then it would be really helpful if these docs could be updated to reflect this. Otherwise I'm left guessing as to what version I should download, what files from other tools need to be available, and in the end I simply had to download multiple setup files until I saw something with the same build number pop up on screen.

Or better yet, possibly consider moving these to a different location please, e.g. on github ;) To be blunt, this has been an insane level to have to go to to get the equivalent of a `dmesg` output for diagnosing a USB device, and if these are just small plaintext utility files for parsing a binary schema, I feel they could be better placed here primarily and then bundled into builds of the kit if needed, rather than the other way around. 

Thanks

mhopkins-msft, 2023-06-21:
Hi @stellarpower,

Thank you for reaching out to us. I'm having a little trouble parsing exactly what you'd like to see changed. The NetMon docs state that it is an archived tool, meaning, it is no longer current. 

Have you checked out the [Event Tracing for Windows (ETW)](https://learn.microsoft.com/windows-hardware/drivers/devtest/event-tracing-for-windows--etw-) section?

Can you tell me what you need to accomplish? Maybe there is some other documentation that would be more helpful.

Thanks,
 ~ *Mark*

mhopkins-msft, 2023-06-23:
Hi @stellarpower,

I'm going to close this issue for now. If you have any specific suggestions about how we can improve the docs to help here, please feel free to re-open this issue.

Thanks again for reaching out to us!
 ~ *Mark*

closed 2023-06-23 by mhopkins-msft

stellarpower, 2023-06-25:
Sorry for the delay!

So I'm trying to follow [this guide](https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/how-to-capture-a-usb-event-trace) on getting logs for USB events - I am trying to debug why Windows is unhappy talking to a hardware USB device, and without having logging back from the OS it's quite a lot of guesswork. I can run logman, so far so good, but then it dumps out an ETL file and that's what I'm trying to view.

I have tried installing [PerfView](https://github.com/microsoft/perfview), and whilst this seemed to be able to parse the ETL files, what it displayed on screen was packed onto one giant line and so almost impossible to read. Also it looks like ETW would need to be incorporated into a custom driver(?) - for my usecase we're not needing to write a custom driver, just want to obtain some logs from userspace when I plug and unplug a USB device to see why Windows isn't too happy with what has gone on. Device manager is giving a pretty generic error so am not sure where else we can get logging information.

The issue for me is that the instructions on how to install the files needed weren't reproducible - I had to download multiple versions of the Windows SDK and the driver development kit until I found the one that contained all of the scripts needed, and even when I did I had to search for the files and install some other subfolders using the same method.

Ideal solution would be:

- As they're only a few hundred KiB plaintext files, upload the files that are packaged into the Windows Driver Kit needed to parse the ETL files to a separate github repo, so that users need not install the full development kit to get access to them just for logging purposes. And then adjust the instructions accordingly to point to this s the download location
- or, if netmon is deprecated, if you could recommend what tool can be used instead to read USB (and/or other) events from an ETL file, and link to how to install and configure this.

And if not, at the least I'd like at least updating to mention exactly which version(s) of the WDK need to be installed, and the code formatting on the "Installing and using netmon" page could probably be reworked as it just shows as plain text to me. If it if possible to locate the installation directory of the WDK programatically, a section that I could just paste into powershell that would perform all steps needed to install the netmon parsers would be nice.

Let me know if you'd still like further clarification :)