MicrosoftDocs/windows-driver-docs #3559: When I open a .pfx file, I get to see it in Notepad. Totally useless. What to do?

MarcSchluperAtIntel, 2023-06-16:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 19d00cd0-55a7-4795-7d7a-fb4dd821177e
* Version Independent ID: d689a270-6fb0-2b14-d368-58813b264d69
* Content: [Importing an SPC into a Certificate Store - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/importing-an-spc-into-a-certificate-store)
* Content Source: [windows-driver-docs-pr/install/importing-an-spc-into-a-certificate-store.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/importing-an-spc-into-a-certificate-store.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2023-06-16:
Hey Marc, thanks for the feedback.  What is it you're trying to do with the .pfx file?

MarcSchluperAtIntel, 2023-06-16:
I was trying to install a web app certificate b/c today it expired (and I did not get a warning email b/c the previous owner did not transfer ownership) and the web application was not accessible anymore. Instant urgency.
The default action of the .pfx file was not to install the certificate but to open it in Notepad.

And I could not find how to change the default action.

Since then I have found certlm.msc (imagine that’s your name – nobody would ever remember it).

In there, I found a certificate Import Wizard

***@***.***

Maybe mentioning this alternative way to install a certificate (on the Microsoft page I was) could help someone else who panicked b/c a simple installation of a certificate was not working for them.

Thanks,
Marc

tedhudek, 2023-06-16:
Thanks for the suggestion Marc, I've added that to the page in question.  Glad you got your issue resolved!

closed 2023-06-16 by tedhudek