MicrosoftDocs/windows-driver-docs #3562: Advice for systems bundled with a limited/old version of pnputil

kamild1996, 2023-06-19:
This page recommends against using DevCon in favor of PnPUtil, explaining that DevCon was never meant to be used as a proper tool for managing drivers.
However, it doesn't go into detail about situations where the PnPUtil tool provided with operating systems as old as the previous generation (such as Windows Server 2019) offers much more limited functionality, rendering most of the provided equivalent commands unavailable. 
While the "[PnPUtil Command Syntax](https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/pnputil-command-syntax)" elaborates further on the availability of these features, I believe this page should also mention this topic, and perhaps offer a different alternative to affected users (if any).

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: a9bdd20b-2b92-daad-5af2-ba3655189c93
* Version Independent ID: a9bdd20b-2b92-daad-5af2-ba3655189c93
* Content: [Replacing Device Console (DevCon.exe) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/devcon-migration)
* Content Source: [windows-driver-docs-pr/devtest/devcon-migration.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/devcon-migration.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

tedhudek, 2023-06-19:
Hi @kamild1996,

Thanks for the feedback, it's an excellent point.

I've added a sentence just before the table at https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/devcon-migration#table-of-equivalencies to link over to the PnPUtil command syntax page.  As you've noticed, it does have the OS version when each feature was added to the tool.  Note that this is not necessarily the earliest version of Windows that a given command will run on.

Is there a specific PnPUtil command you're curious about running on earlier Windows versions?  Or a blocking issue you're experiencing?

I don't know of another alternative, unless you want to use DevCon, which is now officially unsupported and use at your own risk.

tedhudek, 2023-06-21:
Closing out for now, please re-open if still needed

closed 2023-06-21 by tedhudek