MicrosoftDocs/windows-driver-docs #3563: How does APN xml match SIM card of different mobile operators with correct APN profile?

BobXGY, 2023-06-20:
Hi Microsoft experts,

I am developing my tablet(with Qualcomm cellular modem) and I want to make sure that users can access Internet once insert their SIM card.
Due to I have no method to view full APN list of Windows 11, I am not sure that all operators have correct APNs on Windows11.
So, I want to add some APN profiles for specific mobile operators and I find following document, but it is still unclear for me to know how does Windows match APN profile with SIM card, ICCID, MCC MNC or IMSI?
I would be very appreciate If you can improve this document and I will be able to add APN for MOs.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 9ad6c4f7-b1a9-4939-295c-2cb4f743c71e
* Version Independent ID: 7f5261a7-3116-5629-584a-9c6e116c6a49
* Content: [APN example - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/mobilebroadband/apn-example)
* Content Source: [windows-driver-docs-pr/mobilebroadband/apn-example.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/mobilebroadband/apn-example.md)
* Product: **windows-hardware**
* Technology: **mobilebroadband**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2023-06-26:
Hi @BobXGY,
Thanks for opening this issue. 
The COSA database contains the inbox data for Windows, including APNs. 
[Desktop COSA/APN database settings](https://learn.microsoft.com/en-us/windows-hardware/drivers/mobilebroadband/desktop-cosa-apn-database-settings) shows that MCC, MNC, IMSI range, ICCID range, and SPN are used to match the profile.
[Testing your desktop COSA/APN database submission - Windows drivers | Microsoft Learn](https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Flearn.microsoft.com%2Fen-us%2Fwindows-hardware%2Fdrivers%2Fmobilebroadband%2Ftesting-your-desktop-cosa-apn-database-submission&data=05%7C01%7CAmy.Viviano%40microsoft.com%7Cb111db36de5b4dca741408db7677ab6e%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C638234028630485315%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=N6FMvyco0YR%2BBK9tiOt4jQGemZq3wtd5q5Hk3VbkChw%3D&reserved=0) has information for testing the local database if updates are needed, but most of the MOs should be in the Inbox database already.
Does this answer your question?

aviviano, 2023-06-26:
@BobXGY: Additionally, note that operators can decide on which of the identifiers they'd like to choose depending on their use case. While some operators will match devices using MCC/MNC and ICCID ranges, others may use SPN or an IMSI range to match devices with the correct APN profile.

BobXGY, 2023-06-29:
Hi, @aviviano,
Thanks for your reply.
I've tried as this sample to customize APN: https://learn.microsoft.com/zh-cn/windows-hardware/customize/desktop/customize-cosa, 
ppkg has been imported, but target device did not use new APN on Anritsu base station simulator, do you have any debug suggestions?
Additionally, will we able to check complete internal APN list on Windows11? Then we could know if the APN lists are suitable for us.

Best Regards

aviviano, 2023-07-11:
Hi @BobXGY, 
I'm sorry, the documentation team doesn't have the bandwidth to help you with specific debugging questions. You could try posting your question on Stack Overflow, [Microsoft Q&A](https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.microsoft.com%2Fen-us%2Fanswers%2Fproducts%2F&data=04%7C01%7CAmy.Viviano%40microsoft.com%7C8d4e6cff86b74efe1d9208d935d734d2%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637600020691837012%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=o%2F5HRZapf1U05h7Hhi8c4%2B%2FpV02ZOvFSyEe8AVW3f70%3D&reserved=0), or the [OSR NTDEV forum](https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fcommunity.osr.com%2Fcategories%2Fntdev&data=04%7C01%7CAmy.Viviano%40microsoft.com%7C8d4e6cff86b74efe1d9208d935d734d2%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637600020691837012%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=aVALxqj4w%2F%2Fa%2BdLeerBZqIS2SR%2FkJUXJP7MDlrnIPn4%3D&reserved=0).
Also, note that the APN database is gone as of Win 10 v1703. We are working on updating the documentation. 
Thanks again for your questions. Please open a new issue if you have further documentation questions.

closed 2023-07-11 by aviviano