MicrosoftDocs/windows-driver-docs #3567: Link to "Identifiers for Audio Devices" is broken

Sumit-Nbase2, 2023-06-22:
[Enter feedback here]

Hello Microsoft, 

This is a feedback for the page in Description:
Link to "Identifiers for Audio Devices" is broken. 
[Identifiers for Audio Devices](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/identifiers-for-isapnp-devices) is tagged with ISAPNP Device Identifiers.

Please update the link for "Identifiers for Audio Devices". 

Thanks,
Sumit

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: e29e581e-02a0-dab0-83db-25ae1ca1aebb
* Version Independent ID: 21141809-e821-8e58-dcf2-be1d1bb9be87
* Content: [Step 1 The New Device is Identified - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/step-1--the-new-device-is-identified)
* Content Source: [windows-driver-docs-pr/install/step-1--the-new-device-is-identified.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/step-1--the-new-device-is-identified.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

DOMARS, 2023-06-22:
@Sumit-Nbase2 - Thanks for seeing this error on my part and creating this helpful issue. The link is now fixed. Thanks for caring about the docs, and helping the next reader get to the proper audio topic!

closed 2023-06-22 by DOMARS

Sumit-Nbase2, 2023-06-23:
Most Welcome :)