MicrosoftDocs/windows-driver-docs #3574: information about the Microsoft driver signing process

swgide, 2023-06-23:
Dear support for the Microsoft learn pages
I am looking for information about the Microsoft driver signing process.  
Can you please clarify who has to submit the driver package (hlkx; WIN10) to Microsoft for signing.  
Does it have to be company A or can company B, on behalf of company A, also submit the driver package.  
Which Microsoft development account must be used? 
Best Regards

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: a4b96490-ed15-01c6-9cbc-8e357919af3d
* Version Independent ID: a4b96490-ed15-01c6-9cbc-8e357919af3d
* Content: [Getting started with the hardware submission process - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-started-dashboard-submissions)
* Content Source: [windows-driver-docs-pr/dashboard/get-started-dashboard-submissions.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-started-dashboard-submissions.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @anaharris-ms
* Microsoft Alias: **anaharris**

95HenryFong95, 2023-06-23:
Hi . Friend.the PC will go to blue PG then count drown to 100% then gone black no response but sometime as usual need to sign in after that I try to click on the Google chrome icon no response only before it goes black I spotted ...F2. os I try on that page still use less . is there any way to make it alive again. although it's old enough. but it still looks great.  Thank for your time on helping. Good luck.

anaharris-ms, 2023-06-23:
Hello @swgide,

Per our agreements, company A must submit their own drivers. If you have any other specific questions, I suggest you content support: https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-started-dashboard-submissions.

Thank you and have a great day!

Anastasia

closed 2023-06-26 by anaharris-ms