MicrosoftDocs/windows-driver-docs #3582: Update for LE Audio support

guliteb, 2023-06-29:
Microsoft has announced support of Bluetooth LE Audio
https://blogs.windows.com/windowsexperience/2023/05/23/announcing-new-windows-11-innovation-with-features-for-secure-efficient-it-management-and-intuitive-user-experience/

LE Audio is based on version 5.3 of the BT specification.

When will Microsoft update this page to reflect the bluetooth version and profile support?

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 27a0ed90-dff4-f356-1853-de669b0d1540
* Version Independent ID: d846c2ad-9b05-8e61-45f0-19e50734f0f3
* Content: [Bluetooth version and profile support in Windows 11 - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/bluetooth/general-bluetooth-support-in-windows)
* Content Source: [windows-driver-docs-pr/bluetooth/general-bluetooth-support-in-windows.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bluetooth/general-bluetooth-support-in-windows.md)
* Product: **windows-hardware**
* Technology: **bluetooth**
* GitHub Login: @mhopkins-msft
* Microsoft Alias: **mhopkins**

mhopkins-msft, 2023-07-07:
Hi @guliteb,

It's taking longer than expected to get the information I need to update this page. I'll let you know here as soon as it is updated.

Thanks,
 ~ *Mark*

mhopkins-msft, 2023-07-21:
Hi @guliteb,

The article has been updated with the latest Windows 11 Bluetooth support information. Thank you for your patience!

 ~ *Mark*

closed 2023-07-21 by mhopkins-msft