MicrosoftDocs/windows-driver-docs #3586: windows11+intel BT5.3 bluetooth SCO  remote connect Failure

anone313, 2023-07-02:
brb = (struct _BRB_SCO_OPEN_CHANNEL*) &(connection->ConnectDisconnectBrb);
    devCtx->ProfileDrvInterface.BthReuseBrb((PBRB)brb, BRB_SCO_OPEN_CHANNEL_RESPONSE);
	
	brb->Hdr.ClientContext[0]	= connectionObject;
	brb->BtAddress				= ConnectParams->BtAddress;
	brb->ChannelHandle			= ConnectParams->ConnectionHandle;
	brb->Response				= SCO_CONNECT_RSP_RESPONSE_SUCCESS;
	
	brb->TransmitBandwidth		= 8000;
	brb->ReceiveBandwidth		= 8000;  // 64Kb/s
	brb->MaxLatency = 0xFFFF;// 50;

	brb->PacketType = 0x3F;// SCO_EV3;///**/devCtx->ScoPacketTypes;  SCO_HV1 | SCO_HV3
	brb->ContentFormat = SCO_VS_IN_CODING_LINEAR | SCO_VS_IN_SAMPLE_SIZE_16BIT | SCO_VS_AIR_CODING_FORMAT_CVSD; // SCO_VS_SETTING_DEFAULT;//SCO_VS_IN_CODING_LINEAR | SCO_VS_IN_SAMPLE_SIZE_16BIT | SCO_VS_AIR_CODING_FORMAT_CVSD;
	brb->RetransmissionEffort = SCO_RETRANSMISSION_MIN1_QUALITY;// SCO_RETRANSMISSION_MIN1_POWER;

	brb->ChannelFlags			= SCO_CF_LINK_SUPPRESS_PIN;
	brb->CallbackFlags			= CALLBACK_DISCONNECT;
	brb->Callback				= &HpSrvIndicationCallback;
	brb->CallbackContext		= connectionObject;
	brb->ReferenceObject		= (void*) WdfDeviceWdmGetDeviceObject(devCtx->Device);
	status = SendBrbAsync(devCtx->IoTarget, connection->ConnectDisconnectRequest, (PBRB)brb, sizeof(*brb), RemoteConnectCompletion, brb);
	if (!NT_SUCCESS(status))
		TraceEvents(TRACE_LEVEL_ERROR, DBG_CONNECT, "SendBrbAsync failed, Status = %X", status);
		
		
		
The above code can be connected normally before win11 Bluetooth 5.3, but in win11 Bluetooth 5.3 version brb->PacketType and brb->ContentFormat, no matter how you set the return value of status, the return value is always 0xC00000BB(STATUS_NOT_SUPPORTED)

anone313, 2023-07-03:
Initialization uses BRB_SCO_GET_SYSTEM_INFO to obtain SCO information, and the result is:
BRB_SCO_GET_SYSTEM_INFO:
	Features = 3
	MaxChannels = 2
	TransferUnit = 0
	PacketTypes = 3F
	DataFormats = F
From the results, SCO is supported???but why does the SCO connection above always return STATUS_NOT_SUPPORTED?

mhopkins-msft, 2023-07-05:
Hi @anone313

Thank you for reaching out to the driver documentation team. The men and women who staff the driver documentation team are documentation professionals with limited resources and are unable to respond to individual requests for support.

You might try posting your question on the [OSR Developer Community forums](https://community.osr.com/). There are many other Windows system developers there who may be able to assist you.

You could also post your question on the [Microsoft Q&A site](https://learn.microsoft.com/en-us/answers/).

I wish you the best in resolving your questions. If you have a recommendation for updating the documentation, please feel free to reopen this issue.

Thank you,
 ~ *Mark*

closed 2023-07-05 by mhopkins-msft

anone313, 2023-07-10:
hello,@mhopkins-msft 
   Thank you very much
Best regards,  
anone313