MicrosoftDocs/windows-driver-docs #3593: Floating number types are wrong

aleksazr, 2023-07-06:
In section [Using a Memory Window](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/memory-window#using-a-memory-window), 5th bullet: "10-byte, 16-byte, 32-byte, and 64-byte real numbers;"

"32-byte, and 64-byte" is wrong, it should be:

32-bits, 64-bits, 10-byte and 16-byte real numbers;
OR
4-byte, 8-byte, 10-byte and 16-byte real numbers;

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: c1a2991c-fcc9-4d78-2573-0af90284b597
* Version Independent ID: 8aec56bc-62d2-a709-74f1-9a18fd975f55
* Content: [Viewing and Editing Memory in WinDbg (Classic) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/memory-window)
* Content Source: [windows-driver-docs-pr/debugger/memory-window.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/memory-window.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

aleksazr, 2023-07-06:
Also, I would like to add that WinDbg doesn't show Real (16-byte) numbers correctly.

I'm testing a [SoftFloat](http://www.jhauser.us/arithmetic/SoftFloat.html) library, which can use 16-bytes (128-bits) real numbers.
I'm dividing 5 by 3, and the result is 1.666, which is 0x3FFFAAAA AAAAAAAA AAAAAAAA AAAAAAAB.
You can check the hex value [here](http://weitz.de/ieee/) - go to the bottom of page, click "binary128",
then type 1.6666666666666666666666666666666667 in the first field.

But, WinDbg sees that as -1.58537649722266472607e-1644, instead of 1.6

P.S.
I don't know where to report the WinDbg bug, so I'm doing it here.

DOMARS, 2023-07-13:
@aleksazr - Thanks for filing this docs issues, and I updated the topic to reflect the bit options displayed in the classic Windows debugger memory window.

On the displayed value in the memory window, you can file WinDbg issues here.  

http://aka.ms/dexex 

You may consider replicating the issue in the current debugger instead of the classic, and filing the issue for the current debugger:
https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-view-preview

If you are able to replicate the issue using the command line cpp or MASM math operands, such as ? 5/3 that would make it easier to work with.
https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/---evaluate-expression-

You may already be aware that compiler options can change how floating-point numbers are treated. 
https://learn.microsoft.com/en-us/cpp/build/reference/fpcvt?view=msvc-170

And the processor family itself of course will treat floating point numbers differently.
https://www.intel.com/content/dam/develop/external/us/en/documents/floating-point-reference-sheet-v2-13.pdf

You may want to investigate the value stored in the registers by changing the register display using the rm command.
https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/rm--register-mask-

Good luck with your floating-point work, and thanks for your interest in the debugger docs.

closed 2023-07-13 by DOMARS