MicrosoftDocs/windows-driver-docs #3595: Lack of good examples

pbhogale-12166972, 2023-07-12:
[Enter feedback here]
Not enough examples provided for the combination of the DDInstall section for different OS.
No examples provided to justify and explain the syntax: "[install-section-name.ntia64.HW] | (Windows XP and later versions of Windows)"
Kindly provide better examples.
---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: fc8d1bbe-fc08-d3de-4dd7-c4a476f320ba
* Version Independent ID: 784b74f5-0dcd-9a19-bcb5-ff9708707e10
* Content: [INF DDInstall.HW section - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/inf-ddinstall-hw-section)
* Content Source: [windows-driver-docs-pr/install/inf-ddinstall-hw-section.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-ddinstall-hw-section.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

pbhogale-12166972, 2023-07-12:
https://learn.microsoft.com/en-us/windows-hardware/drivers/install/inf-manufacturer-section provides good examples on how to use options for different OS Builds. Similar well structured and detailed examples needed for the DDInstall section for various OS Builds.

tedhudek, 2023-07-12:
Hi @pbhogale-12166972, for this one, does the info in https://learn.microsoft.com/en-us/windows-hardware/drivers/install/creating-inf-files-for-multiple-platforms-and-operating-systems answer your questions?

pbhogale-12166972, 2023-07-12:
Thank you for the quick response. I still don't see the example for combining DDInstall section with various OS build options.
I really am looking for an example for the syntax "**[install-section-name.ntia64.HW] | (Windows XP and later versions of Windows)**"
So an example for DDInstall for Windows OS with build 19044, and another DDInstall for Windows OS with build 22000.
I used the "https://learn.microsoft.com/en-us/windows-hardware/drivers/install/creating-inf-files-for-multiple-platforms-and-operating-systems" for creating OS dependent Manufacturer section, but I am not able to duplicate this syntax for DDInstall section.

pbhogale-12166972, 2023-07-12:
Can you please help me with such a scenario?

**[Example_DDInstall.HW] ; Empty section, so this INF does not support
.                  ; NT 10.0..19044 and later, but not for NT 10.0..22000 and later
.
[Example_DDInstall.HW]; Used for NT 10.0..22000 and later
.                  ; **

tedhudek, 2023-07-12:
There is no *targetosversion* decoration for **DDInstall** or **DDInstall.HW**.

*targetosversion* is just for Models sections.  To get different installation behavior based on OS version, fork at the Models section.  Then the Models section for the different OS versions should point at differently named DDInstall/DDInstall.HW sections.

Does that make sense?

pbhogale-12166972, 2023-07-12:
Understood, this makes sense and connects the dots for me now. Thanks a lot @tedhudek . Appreciate the quick help.

closed 2023-07-12 by tedhudek