MicrosoftDocs/windows-driver-docs #3600: need to reformulate

NissimBendanan, 2023-07-17:
The following sentence is not understandable, please formulate it in other words:
"The Windows 11, version 22H2 Software Development Kit (SDK) that is compatible with the Windows 11, version 22H2 WDK is not included in Visual Studio. Please use the SDK download link in step 2 below."

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* Technology: **whats-new-and-downloads**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2023-07-17:
Hi @NissimBendanan, thanks for the feedback.  You're right, the old language was awkward.  I've updated the text in steps 1 and 2.  See what you think!

NissimBendanan, 2023-07-18:
thank you, the new text is much better understandable

closed 2023-07-18 by tedhudek