MicrosoftDocs/windows-driver-docs #3602: WinDbg Linux debugging documentation completely missing

exoosh, 2023-07-18:
Hi there,

I learned not too long ago that WinDbgX understands core files written by .NET on Linux in native/managed debugging scenarios.

For a post-mortem we were trying to tease out information from WinDbg regarding the call stack that led to a crash.

And indeed by setting the symbol path to find the shared object itself, `kb` would now proclaim to know that one of the stack frames was happening at a method `start+0x105`.

Alas, even though the file (a native shared object) is not stripped, WinDbg is unable to correlate this to a source line or so.

I have seen this with Windows debugging before whenever symbols were totally unavailable. In this case WinDbg would fall back to referring to the locations by some exported symbol plus and offset.

During my search I also tried looking up more details and came up empty-handed.

Could someone here point me to documentation related to debugging -- or specifically post-mortem debugging scenarios -- for Linux using WinDbgX?!

I am also interested in learning whether the compiler/linker commands should/could be tweaked to yield better results (thinking `-ggdb -g3` and friends).

Any pointers to existing documentation would be appreciated. Otherwise, could you please add it? I know it's still called **Win**Dbg, but seemingly it can do a whole lot more by now.

DOMARS, 2023-07-21:
@exoosh - Thanks for this input on adding more information on the use of Symbols in a Linux environment. We use input such as this, to guide the docs work that we do.  

No matter the OS, with the local symbol cache that WinDbg uses, and the many options, there is some complexity in using symbols. These two topics provide general information on that: 
- [Setting Symbol and Executable Image Paths in WinDbg](https://learn.microsoft.com/windows-hardware/drivers/debugger/setting-symbol-and-source-paths-in-windbg)
- [Symbol path for Windows debuggers](https://learn.microsoft.com/windows-hardware/drivers/debugger/symbol-path)

In version  1.0.1908.30002, Support for Open Enclave debugging was added, for more information, see [Open Enclave debugging](open-enclave-debugging.md). Also as part of supporting Open Enclave, WinDbg can open ELF core dumps and binaries as well as DWARF symbols (DWARF 5 is not currently supported) from both Enclaves and Linux applications. When opening a core dump from a non-Windows application, basic windows and commands should all work properly, but most extensions and Windows-specific commands will not work. ELF and DWARF files will be downloaded from symbol servers following the [key conventions defined here](https://github.com/dotnet/symstore/blob/master/docs/specs/SSQP_Key_Conventions.md). Enclaves are the only supported scenario.

Starting with version 1.0.2007.01003 of the Windows Debugger, Portable PDB Symbols are supported. This is described in this topic, and it also provides links to related ECMA specifications. [Portable PDB Symbols](https://learn.microsoft.com/windows-hardware/drivers/debugger/symbols-portable-pdb).

Symbols are a great help, but in all settings, you can almost always read registers and disassemble instructions, but that is tedious and not a great option, and assumes some knowledge of assembler.

If you have general feedback on how WinDbg can better support Linux, you can provide that using the WinDbg issues repo - http://aka.ms/dexex 

Thanks again for your interest in WinDbg and the input on the docs.

closed 2023-07-21 by DOMARS

exoosh, 2023-07-24:
Thanks @DOMARS for the comment and the pointers. This helps a lot already.

> When opening a core dump from a non-Windows application, basic windows and commands should all work properly, but most extensions and Windows-specific commands will not work.

Sure, makes sense. I wouldn't expect anything else.

> ELF and DWARF files will be downloaded from symbol servers following the [key conventions defined here](https://github.com/dotnet/symstore/blob/master/docs/specs/SSQP_Key_Conventions.md). Enclaves are the only supported scenario.

Perhaps you could clarify a few things.

Enclaves being the only _supported_ scenario precludes other ELF files or means they might (or even should) work, but it's not supported if I'd turn to the Microsoft support?

Is it limited to outright symbol servers, or are local symbol stores fine?

> Starting with version 1.0.2007.01003 of the Windows Debugger, [Portable PDB Symbols](https://learn.microsoft.com/windows-hardware/drivers/debugger/symbols-portable-pdb) are supported. This is described in this topic, and it also provides links to related ECMA specifications. Portable PDB Symbols.

I was aware of portable PDBs, but the issue we encounter is in native code and that was the blind spot to me.

Reading assembly is no issue, however.

Thanks again for all the valuable pointers.