MicrosoftDocs/windows-driver-docs #3603: WinDbg Preview Please document how to set environment variables for Start Debugging > Launch executable

buzz3791, 2023-07-18:
Please document how to set environment variables for WinDbg Preview version 1.2306.12001.0 Start Debugging > Launch executable

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: ee388ff3-b1a4-1100-6026-bfc6c0ca72e8
* Version Independent ID: 3fae1e70-4d43-6a72-2d49-db00a0607fbc
* Content: [Setting Symbol and Executable Image Paths in WinDbg - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/setting-symbol-and-source-paths-in-windbg)
* Content Source: [windows-driver-docs-pr/debugger/setting-symbol-and-source-paths-in-windbg.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/setting-symbol-and-source-paths-in-windbg.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2023-07-21:
@buzz3791 - You can set symbol behavior on load for WinDbg, that is described here:
https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-command-line-preview

I would also look at Workspace settings, which persist and can cause confusion.
https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-setup-preview

I tend to use the !symfix command, but that may not be enough depending on your setup. I would use the .sympath command to display your active path and [!sym noisy](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/-sym) to monitor symbol loading. 

With the local symbol cache that WinDbg uses, and the many options to set it, there is some complexity in using symbols. This topic provides information on that: 

- [Symbol path for Windows debuggers](https://learn.microsoft.com/windows-hardware/drivers/debugger/symbol-path)

The echo driver lab shows how to combine local symbols from a Visual Studio project into the OS symbols provided on the Microsoft Symbol server.
- [Debug Windows drivers step-by-step lab (echo kernel mode)](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/debug-universal-drivers---step-by-step-lab--echo-kernel-mode-)

For additional questions on the multiple ways on setting and working with the symbol path, I would recommend the OSR debugging forum: 
- [OSR's windbg List](https://community.osr.com/categories/windbg)

closed 2023-07-21 by DOMARS

buzz3791, 2023-07-22:
@DOMARS Hi Don, thanks for your reply.  Please re-open this issue because your reply did not address my issue.  I'm not asking about symbol .pdb files, I'm asking about supporting values retrieve via std::getenv().

I'm asking for documentation as follows:
Lets say I have a C++ app myapp.exe that contains code like:
#include <iostream>
#include <cstdlib>
int main()
{
    if(const char* env_var1 = std::getenv("MYVAR1"))
        [std::cout](http://en.cppreference.com/w/cpp/io/cout) << "Your MYVAR1 is: " << env_var1 << '\n';
}

I'm asking for information on how to configure the launch of myapp.exe so that I can set a value for the MYVAR1 environment variable.  When launching from a cmd.exe prompt I would do the following:

set MYVAR1 = "myvalue1"
myapp.exe

Then at runtime MYVAR1 would be myvalue1.

How can I achieve this when using WinDBG to launch myapp.exe?  Note that Visual Studio has this feature [1].

1 - "Environment row" in the table titled "Configuration Properties folder (Debugging category)" - https://learn.microsoft.com/en-us/visualstudio/debugger/project-settings-for-a-cpp-debug-configuration?view=vs-2022