MicrosoftDocs/windows-driver-docs #3607: "PurposeGroups" in Desktop COSA-only settings

BobXGY, 2023-07-25:
Hi, Microsoft expert.

About this setting "PurposeGroups", the document page shows "A string specifying the purposes of the connection by a comma-separated list of GUIDs representing purpose values." And the note as following, but I didn't see any corresponding GUIDs.

> The following purpose values are available:
> - Internet
> - MMS
> - IMS
> - SUPL
> - Purchase
> - Administrative
> - Application
> - eSIM provisioning

Can you please provide the corresponding GUIDs for configuration?

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: a676511c-bfcf-502d-4df9-ded1957298d3
* Version Independent ID: 5e04616e-9968-b8be-44cd-074307c7aa72
* Content: [Desktop COSA/APN database settings - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/mobilebroadband/desktop-cosa-apn-database-settings)
* Content Source: [windows-driver-docs-pr/mobilebroadband/desktop-cosa-apn-database-settings.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/mobilebroadband/desktop-cosa-apn-database-settings.md)
* Product: **windows-hardware**
* Technology: **mobilebroadband**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2023-12-20:
Hi @BobXGY,
Thanks for opening this issue. Were you able to resolve this, or are you still looking for the GUIDs?

aviviano, 2024-01-02:
@BobXGY - Note that we are currently working on updates for this page and will close this issue for now. If you are still having trouble, please reopen this issue or open a new one. 
Thanks again.

closed 2024-01-02 by aviviano