MicrosoftDocs/windows-driver-docs #3609: I want to know status enabled or disabled currently

kdm6389, 2023-07-28:
I want to know status enabled or disabled currently, how can I do this using PnPUtil

[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 41c58204-8ae9-7822-df54-6e33af08f3c7
* Version Independent ID: 3ded15a1-1b56-e571-adfd-383b27f64217
* Content: [PnPUtil Examples - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/pnputil-examples)
* Content Source: [windows-driver-docs-pr/devtest/pnputil-examples.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/pnputil-examples.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

tedhudek, 2023-07-31:
Hi @kdm6389, thanks for the question.  Basically you'll want to use `pnputil /enum-devices`.  I just updated https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/pnputil-examples to clarify and also to alphabetize the examples so they're easier to find.  Hope this helps!

closed 2023-07-31 by tedhudek

kdm6389, 2023-08-15:
pnputil /enum-devices woked thanks