MicrosoftDocs/windows-driver-docs #3610: Can't add new certificate

jbc22, 2023-08-04:
Problem:

When following the steps to add a New Certificate to the organization, it fails with a message of "We encountered a problem processing the file you uploaded. please ensure the file you sign and upload is the same file you downloaded."

The test SignableFile.bin passes signtool.com verify /pa and DigiCert's Certificate Utility verification.

[image]

I am using signtool v10.0.19041.685 with the arguments "/tr http://timestamp.digicert.com /fd sha256 /td sha256".

Chrome Version 115.0.5790.110 (Official Build) (64-bit) on Windows 11 Home 22621.1992.

I have tried using Edge on the same PC, as well as Chrome on MacOS.

Side problem:

I'm submitting this in case it's related. When trying to remove old certificates, it fails and presents an "[object][object]" message.
[image]

tedhudek, 2023-08-04:
Hi @jbc22 would you happen to have the URL of the page with the steps you are referencing?

jbc22, 2023-08-05:
@tedhudek @anaharris-ms the URL: https://partner.microsoft.com/en-us/dashboard/account/v3/managecertificates?redirectedBy=AFD . You can confirm that's the URL I'm using in the screenshot above.

That page is responsible for both the main problem (addition of certificates) and the side problem (removal of certificates).

anaharris-ms, 2023-08-07:
Hi @jbc22 ,

Have you attempted to contact support from that page?

jbc22, 2023-08-07:
@anaharris-ms I have, without much luck.

jbc22, 2023-08-08:
@anaharris-ms @tedhudek I'm not sure where I can reach out for help on this. Any guidance in the right direction would be much appreciated!

spatil-intel, 2023-08-17:
Can you try using SHA384 instead of SHA256 for the signature ? It seems like Microsoft updated partner center recently and seem to be rejecting SHA256 signatures.  They need to update their documentation : https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-cert-manage

spatil-intel, 2023-08-18:
We just got confirmation that Microsoft has fixed the issue of not accepting SHA256 signatures for SignableFile.bin. Looks like it is working again the way it used too. The issue with removal of certificates still remains though.

closed 2023-10-25 by tedhudek