MicrosoftDocs/windows-driver-docs #3613: Please add description for param 1 = 0x12

jalpapat, 2023-08-09:
Looking for more info what does 0x12 mean. Thanks.
*** Fatal System Error: 0x00000119
                       (0x0000000000000012,0xFFFFC20133065410,0x0000000000000000,0xFFFFFFFFFFFFFFFF)

 

Break instruction exception - code 80000003 (first chance)

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 7917857b-97bc-efec-67ab-5e3af061524c
* Version Independent ID: 8c00d20e-9b21-7bb0-a041-220aca9d2487
* Content: [Bug Check 0x119 VIDEO_SCHEDULER_INTERNAL_ERROR - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x119---video-scheduler-internal-error)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x119---video-scheduler-internal-error.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x119---video-scheduler-internal-error.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2023-08-18:
@jalpapat - I added information on the x12 parameter. It indicates that the video driver did not manage a native "fenced" area of memory properly. With the sub parameters like this that point to the video driver, the normal recommendation is to check with the video chip / card vendor. To both communicate that their driver is causing this type of bug check, and to see if they have an updated driver available. I hope you are able to find a resolution to this bug check on your PC, and thanks for filing this PR requesting that we add information of x12.

closed 2023-08-18 by DOMARS

jalpapat, 2023-08-18:
Thank you Don for the update and response. Appreciated! I work for Nvidia
and hit an issue with 0x12 parm internally.

Regards,
Jalpa