MicrosoftDocs/windows-driver-docs #3620: SAL annotation incorrectly showing as italic

brinez-msft, 2023-08-24:
SAL 2.0 annotations have a leading and trailing underscore.
However, the online documentation is treating this as markdown formatting and turning them into italics.
Proposed solution: Escape the underscore to that markdown doesn't treat it as formatting.
Alternative solution: Wrap all the SAL annotations in a preformatted code block.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 892e5e70-d246-8263-b90e-477b10cb2fef
* Version Independent ID: a0b24b08-1b25-8304-eeea-2b93b95ae35c
* Content: [SAL 2.0 Annotations for Windows Drivers - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/sal-2-annotations-for-windows-drivers)
* Content Source: [windows-driver-docs-pr/devtest/sal-2-annotations-for-windows-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/sal-2-annotations-for-windows-drivers.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2023-08-24:
@brinez-msft - Thanks for pointing out that our current content engine is parsing it like this. I may also make use of the markdown code literal to fix this - `.

\_IRQL\_is\_cancel\_ 

or

`_IRQL_is_cancel_`

DOMARS, 2023-08-26:
@brinez-msft - I reviewed and believe that I fixed all of the SAL annotations to correctly render the underscores properly. I also converted the HTML to pure markdown which should make this content more resilient to future rendering changes. Thanks for seeing this problem and filing this issue.

closed 2023-08-26 by DOMARS