MicrosoftDocs/windows-driver-docs #3624: Great technical doc. But how can the average person get AEC enabled in Windows 11

jlpetz, 2023-08-30:
Hi Windows Audio Group,

I have a Audio technica AT2020-USBX hooked up to a Windows 11 machine.

I appears the device is recognised and setup automatically using the built in Microsoft USB Audio Driver. From looking at the driver in device manager, it uses the USB Audio Class 1 driver. Not the newer USB class 2.0 driver described here https://learn.microsoft.com/en-us/windows-hardware/drivers/audio/usb-2-0-audio-drivers

Regardless. I'd like to be able to enable at the Operating System level, a software based AEC and noise reduction effect on the microphone. I work from home and rather than wear headphones all day, I use speakers (seperate USB Audio device) with the microphone. This is generally not a big problem as most conferencing application nowdays have their own noise+echo/AEC implementations. But these implementation are often working based off just the meeting audio for the reference stream, meaning some OS sounds creap into the microphone audio still. Also this doesn't work consistenly across the entire OS.

I'm reaching out here, as I'm not sure where else to ask this. But since Microsoft has clearly done a lot of work to provide a generic USB Audio class 1.0 & 2.0 driver and a framwork for APOs including AEC. Could you combine this for work customers and make it possible for us to easily enable AEC and noise reduction for these generic USB Audio driver devices? When I look at properties on the USB Audio device, There is currently no "Enhancement" tab which for many sound card devices (with custom driver) is where you find AEC and noise option. Providing a way to enable a software based noise reduction and AEC to  any audio devices easily, would be a great feature for windows.

Regards Jarrod

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: a4e81f7e-81ed-1d7a-bded-cbef61fcdae1
* Version Independent ID: a4e81f7e-81ed-1d7a-bded-cbef61fcdae1
* Content: [Windows 11 APIs for Audio Processing Objects - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/audio/windows-11-apis-for-audio-processing-objects#reuse-of-the-wm-sfx-and-mfx-apos-in-windows-11-version-22h2)
* Content Source: [windows-driver-docs-pr/audio/windows-11-apis-for-audio-processing-objects.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/windows-11-apis-for-audio-processing-objects.md)
* Product: **windows-hardware**
* Technology: **audio**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

jlpetz, 2023-08-30:
[image]
[image]

DOMARS, 2023-08-31:
@jlpetz - The best route to provide input on the Windows audio experience, such as including audio processing features like automatic gain and echo cancelation, is to use the Windows feedback app as described here:

https://support.microsoft.com/windows/send-feedback-to-microsoft-with-the-feedback-hub-app-f59187f8-8739-22d6-ba93-f66612949332

You can see if others have offered similar input, and/or create a new feedback item.

It looks like you are seeing the settings, you can always run mmsys.cpl to get to the audio properties to see how things are configured on a specific PC and device combination. Each hardware OEM and external audio device vendor has the opportunity to configure audio effects, to allow unique audio solutions, such as for professional audio devices or microphone arrays.

For more information on how the top layer apps take advantage of audio effects and the different versions of USB audio, the Matthew Vaneerde blog is a good resource.

https://matthewvaneerde.wordpress.com/2019/04/19/how-can-my-usb-audio-device-tell-the-audio-driver-what-kind-of-device-it-is/

Thanks for your interest in improving the Windows audio experience.

closed 2023-08-31 by DOMARS