MicrosoftDocs/windows-driver-docs #3626: Can I use an EV certificate from a private trusted CA to sign my driver for attestation signing ?

PJ-Dad, 2023-08-31:
**Can I use an EV certificate from a private trusted CA to sign my driver for the attestation method of signing ?** 

The private CA is the company I work for and they are authorised as a CA by Entrust.

Only public Certificate Authorities (CA) are listed on this page, with no indication if MS will reject a driver signed with an EV certificate fro ma private CA for attestation signing.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 0810e3cb-a0c4-7743-39c4-6a7ac70dcccc
* Version Independent ID: 0810e3cb-a0c4-7743-39c4-6a7ac70dcccc
* Content: [Driver code signing requirements - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-reqs#windows-10-attestation-signed-drivers-for-testing-scenarios)
* Content Source: [windows-driver-docs-pr/dashboard/code-signing-reqs.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/code-signing-reqs.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @anaharris-ms
* Microsoft Alias: **anaharris**

PJ-Dad, 2023-09-07:
Hi @anaharris-ms ,  do you have any thoughts on my question ?
thanks
PJ

anaharris-ms, 2023-09-07:
@PJ-Dad,

Sorry for the delayed response. We allow EV certificates from members of the Microsoft Trusted Root Program. Details and requirements can be found here: [Program Requirements - Microsoft Trusted Root Program | Microsoft Learn](https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Flearn.microsoft.com%2Fen-us%2Fsecurity%2Ftrusted-root%2Fprogram-requirements&data=05%7C01%7Canaharris%40microsoft.com%7C1ad05313183644b4067308dbafb3ea85%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C638296959499201964%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=vQ%2FxOVIhu3fhRzKSqLVnxlb3hytfp%2FM%2FwWutKqBjaxM%3D&reserved=0)

PJ-Dad, 2023-09-08:
Thank you very much for answering my question @anaharris-ms .
egards
PJ

closed 2023-09-08 by PJ-Dad