MicrosoftDocs/windows-driver-docs #3631: All four cross-certificates in the list are expired

rcpao-smartm-com, 2023-09-06:
All four cross-certificates in the list are expired.

[image]

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 04c19d51-a948-b3ff-65c2-d8c54291b42f
* Version Independent ID: b5654182-fdcb-72c0-e5ae-536992731e98
* Content: [Cross-Certificates for Kernel Mode Code Signing - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/cross-certificates-for-kernel-mode-code-signing)
* Content Source: [windows-driver-docs-pr/install/cross-certificates-for-kernel-mode-code-signing.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/cross-certificates-for-kernel-mode-code-signing.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

rcpao-smartm-com, 2023-09-06:
Note
The [Microsoft Trusted Root Program](https://learn.microsoft.com/en-us/security/trusted-root/program-requirements) no longer supports root certificates that have kernel mode signing capabilities. For more info, see [Deprecation of Software Publisher Certificates, Commercial Release Certificates, and Commercial Test Certificates](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates).

tedhudek, 2023-09-21:
Thank you for the feedback @rcpao-smartm-com and apologies for the delay.

I have updated https://learn.microsoft.com/en-us/windows-hardware/drivers/install/cross-certificates-for-kernel-mode-code-signing to reflect current reality.

Thanks again for reaching out!

closed 2023-09-21 by tedhudek