MicrosoftDocs/windows-driver-docs #3632: Instructions for looking up HWID are wrong

ppoirier1, 2023-09-06:
This section is outdated and no longer works for finding the HWID -

https://learn.microsoft.com/en-us/windows-hardware/drivers/devapps/print-support-app-association#set-up-printers-and-find-the-hardware-id-or-compatible-id

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: cce233db-310f-5eda-3f9a-676a6b7c60af
* Version Independent ID: 42521d92-86d5-1fc5-2049-8abd741db835
* Content: [Print support app association - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/devapps/print-support-app-association)
* Content Source: [windows-driver-docs-pr/devapps/print-support-app-association.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devapps/print-support-app-association.md)
* Product: **windows-hardware**
* Technology: **devapps**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

lapaz17, 2023-11-27:
@ppoirier1 what is the current method of finding it?

ppoirier1, 2023-11-27:
@lapaz17, for Windows 11:

1. Launch control panel
2. Click "View devices and printers"
3. Click "Printers & scanners"
4. Select a printer
5. Click "Hardware properties"
6. Select the settings tab
7. Click "Devices and Printers Folder"
8. Now continue with step 3 of the referenced Windows 10 instruction list

barrygolden, 2023-12-06:
@ppoirier1, thanks for filing an issue with the Windows Driver Developer Documentation team.

The instructions for finding IPP printer Hardware IDs have been updated to reflect the current user interface in Windows 11, version 23H2.