MicrosoftDocs/windows-driver-docs #3636: Document where the WMIInterface security descriptors are stored and cleanup process

forderud, 2023-09-08:
Can you please extend [INF DDInstall.WMI section](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/inf-ddinstall-wmi-section) with the following:
* Document that the security descriptors are stored in `HKLM\SYSTEM\CurrentControlSet\Control\WMI\Security\<GUID>`
* Document how to clean them up, since uninstalling the driver doesn't seem to delete the registry entries. 

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: f8d1b3a5-9bf4-6158-9552-f43a3d560094
* Version Independent ID: 89c32e23-95d3-09da-f748-892db36d1bef
* Content: [INF DDInstall.WMI section - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/inf-ddinstall-wmi-section)
* Content Source: [windows-driver-docs-pr/install/inf-ddinstall-wmi-section.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-ddinstall-wmi-section.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2023-09-13:
Hello @forderud, thanks for the feedback.

So when you say uninstalling the driver, do you mean uninstalling the device or do you mean removing the driver package from the system?

forderud, 2023-09-14:
> So when you say uninstalling the driver, do you mean uninstalling the device or do you mean removing the driver package from the system?

Sorry for the ambiguity. I actually meant first uninstalling the device and afterwards removing the driver package.

tedhudek, 2023-09-14:
Thanks for the clarification.  Does removing the driver package result in the registry entries being deleted?

forderud, 2023-09-15:
> Thanks for the clarification. Does removing the driver package result in the registry entries being deleted?

No. The WMI security descriptors in `HKLM\SYSTEM\CurrentControlSet\Control\WMI\Security` are still present after running `PNPUTIL /delete-driver <driver-name>.inf /uninstall /force /reboot`. I just tested in Win11 21H2 (OS build 22000.2416).

Robo210, 2023-09-21:
@forderud Is the GUID you are adding security descriptors for an ETW provider/session, or is it an actual WMI interface?

forderud, 2023-09-21:
> @forderud Is the GUID you are adding security descriptors for an ETW provider/session, or is it an actual WMI interface?

The GUID is for an actual WMI interface. Example: [TailLight.inx](https://github.com/forderud/IntelliMouseDriver/blob/main/TailLight/TailLight.inx#L65) that's associated with the GUID in [TailLight.mof](https://github.com/forderud/IntelliMouseDriver/blob/main/TailLight/TailLight.mof#L11).

tedhudek, 2023-09-21:
Hi Fredrik,

Thanks for the feedback and apologies for the delay.  We are tracking the cleanup issue internally - thank you for raising this.  We're not planning any doc updates related to this for now, so I'm going to close out the issue.

We really do appreciate your feedback!

Ted

closed 2023-09-21 by tedhudek

forderud, 2023-09-21:
> Thanks for the feedback and apologies for the delay. We are tracking the cleanup issue internally - thank you for raising this. We're not planning any doc updates related to this for now, so I'm going to close out the issue.

Does this mean that you've reproduced my findings and are working on a root-cause fix?