MicrosoftDocs/windows-driver-docs #3637: where is the lastest wdfcoinstaller?

aspist, 2023-09-09:
[Enter feedback here]
I installed  sdk10.22621 and wdk 10.22621 on windows 11 21H2,but I didn't find the wdfcoinstaller01033.dll in “%program files%\Windows Kits\<version>\redist\wdf” .There is only wdfcoinstaller01009.dll in c:\windows\system.In my other computer which installed windows 7,there is wdfcoinstaller01011.dll in c:\windows\system.The version of wdfcoinstall in windows 7 is later than in windows 11，why？
 
---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: ad07ea10-b00f-48cd-534d-b1ad1bfaa44b
* Version Independent ID: 41c511f9-54b4-07df-d0a5-c2f547fdf293
* Content: [Redistributable Framework Components - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/wdf/installation-components-for-kmdf-drivers)
* Content Source: [windows-driver-docs-pr/wdf/installation-components-for-kmdf-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/wdf/installation-components-for-kmdf-drivers.md)
* Product: **windows-hardware**
* Technology: **wdf**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2023-09-13:
Hi @aspist,

Good question! The WDF versions are listed in a table on https://learn.microsoft.com/en-us/windows-hardware/drivers/wdf/kmdf-version-history.  Windows 7 shipped with version 1.9 of WDF, so it is possible you updated your Windows 7 machine to the 1.11 version.  I am guessing that 22621 (Windows 11, version 22H2) has an old co-installer because co-installers are no longer supported.

You can find more info about WDF co-installer support at https://learn.microsoft.com/en-us/windows-hardware/drivers/wdf/installing-the-framework-s-co-installer.

Does that help?

tedhudek, 2023-09-14:
Closing this out for now, feel free to reopen if still needed.

closed 2023-09-14 by tedhudek