MicrosoftDocs/windows-driver-docs #3639: Could do with a troubleshooting page

rodwiddowson, 2023-09-11:
Apologies to @lorihollasch , @tedhudek  and @cgallred before I even start here,

Some trouble shooting here might be helpful.  I'll bottom this one out (always do) but for instance I am currently looking at `FILTER VERIFIER ERROR:   Filter verifier is in invalid state.` just because I added `ntoskrnl` `ntosext` and `ntfs` to the verified mix.  I don't expect as details guide but a "where to go next" might help,

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 73c29d6f-4cb9-de7f-e233-c5d2b2300f9a
* Version Independent ID: 73c29d6f-4cb9-de7f-e233-c5d2b2300f9a
* Content: [File System Filter Verification - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/file-system-filter-verification)
* Content Source: [windows-driver-docs-pr/devtest/file-system-filter-verification.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/file-system-filter-verification.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

rodwiddowson, 2023-09-11:
PS feel free to close with a "no it doesn't need one".  I promise I won't be offended - it is a bit of a cheek asking here anyway

cgallred, 2023-09-11:
@rodwiddowson this is an error you should not see. Any troubleshooting would be on our end. :-)

This is essentially an internal verifier error that means one of:
* Filter verifier is not enabled yet somehow you got to a point where it should be enabled.
* Verifier callback thunking hasn't happened for your filter.

Could you provide the call stack where this is happening, and perhaps repro steps?

rodwiddowson, 2023-09-12:
Thanks @cgallred 

**Environment**

Windows 11
`Microsoft Windows [Version 10.0.22621.2134]`

```
3: kd> lm vm fltmgr
Browse full module list
start             end                 module name
fffff801`323a0000 fffff801`32416000   FLTMGR     (pdb symbols)
    Loaded symbol image file: FLTMGR.SYS
    Image path: FLTMGR.SYS
    Image name: FLTMGR.SYS
    Browse all global symbols  functions  data
    Image was built with /Brepro flag.
    Timestamp:        B44461FB (This is a reproducible build file hash, not a timestamp)
    CheckSum:         0007B967
    ImageSize:        00076000
    Translations:     0000.04b0 0000.04e4 0409.04b0 0409.04e4
    Information from resource tables:
```

**Stack**
```
FLTMGR!FltpvPrintErrors+0x1d1
FLTMGR!FltpvValidateVerifierState+0x85
FLTMGR!FltvInstanceSetup+0x38
FLTMGR!FltpDoInstanceSetupNotification+0x7c
FLTMGR!FltpInitInstance+0x490
FLTMGR!FltpCreateInstanceFromName+0x1e0
FLTMGR!FltpEnumerateRegistryInstances+0xe3
FLTMGR!FltpDoVolumeNotificationForNewFilter+0xa5
FLTMGR!FltStartFiltering+0x2b
```
**Repro**
It suffices to have ntoskrnl and my drivers 
/verified.

**Other**

Note that I then start getting false positives on ECP release.

I am unable to provide any more information in this forum.  Please reach out to me by mail (@lorihollasch  has it if you don't) and we'll discuss what else I can do.

rodwiddowson, 2023-09-12:
I'll observe that this doesn't happen on Win10 `Microsoft Windows [Version 10.0.19045.3208]` and that this has nothing to do with documentation.  

I was going to close this but I'll leave it to the fine people in documentation to see if a "troubleshooting section" is needed,  I appreciate that there is a fine line here - you don't want to have to tell people what every single assert means because mostrly the text is self-explanatory.

cgallred, 2023-09-13:
This issue (verifier fails when on ntoskrnl and minifilters at the same time) sounded familiar so I did some digging. Turns out that there was a change somewhere along the line that broke having both the kernel and other drivers under verifier simultaneously. If you take out `ntoskrnl` it ought to work.

This should have been fixed starting in build 25123. If you install a recent dev or canary build this issue should no longer repro.

rodwiddowson, 2023-09-14:
Thanks @cgallred i appreciate the info.

DOMARS, 2023-09-19:
@rodwiddowson - I added the following note to the top level driver verifier topic - https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/driver-verifier. 

When using driver verifier with Windows versions 20150 to 25126, if you have selected *ntoskrnl* you may receive an invalid state error.  To avoid this issue either unselect *ntoskrnl* or upgrade to a version of Windows after build 25126.

Longer term, I have thought about a driver verifier walk-through topic, and if I do that work, I will also think about a troubleshooting section. We do mention the [!verifier](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/-verifier) debugger extension in [Monitoring Driver Verifier](https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/monitoring-driver-verifier), but it has been a while since I looked at that, to see what troubleshooting help it might provide. 

Thanks for passing on what you ran into here with DV, and your support of the driver docs and larger driver dev community.

closed 2023-09-19 by DOMARS

rodwiddowson, 2024-02-05:
@DOMARS @cgallred Wow!  

No action needed but I note that Win11 22H3 is still failing because it is still behind 25126.

```Version 10.0.22631.3007```

I have a machine which is enrolled in the early adopters thingy....