MicrosoftDocs/windows-driver-docs #3641: Problems replacing "devcon install" with devgen

forderud, 2023-09-15:
I'm experiencing problems replacing `devcon install` with `devgen`/`pnputil` as described in [Replacing Device Console (DevCon.exe)](https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/devcon-migration) for the [firefly](https://github.com/microsoft/Windows-driver-samples/tree/main/hid/firefly) HID filter-driver sample.

The following "devcon" command will install and activate the driver for a IntelliMouse Explorer 3:  `devcon.exe /r install firefly.inf "HID\Vid_045E&Pid_0047"` . This works nicely and inserts the driver to the device tree as expected.

However, converting the command to devgen/pnputil yields the following result:
```
> devgen.exe /add /bus ROOT /hardwareid "HID\Vid_045E&Pid_0047"
Microsoft Device Generator

Device successfully created. Device Instance ID: ROOT\DEVGEN\{0537181F-703E-5A4A-A90D-9E8F09E3D92E}

> pnputil /add-driver firefly.inf /install /reboot
Microsoft PnP Utility

Adding driver package:  firefly.inf
Driver package added successfully.
Published Name:         oem50.inf
Driver package is up-to-date on device: ROOT\DEVGEN\{0537181F-703E-5A4A-A90D-9E8F09E3D92E}
Driver package is up-to-date on device: HID\VID_045E&PID_0047\6&7d699b6&0&0000

Total driver packages:  1
Added driver packages:  0
```
Removal of the devgen `/hardwareid "HID\Vid_045E&Pid_0047"` argument leads to the same result of the driver package _not_ being installed.

I suspect that the problem is related to the driver in question being a filter-driver. It would be nice if the documentation could be updated to either explain that devcon cannot be replaced with debgen/pnputil for this use-case, or with instructions that also apply to this use-case.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: a9bdd20b-2b92-daad-5af2-ba3655189c93
* Version Independent ID: a9bdd20b-2b92-daad-5af2-ba3655189c93
* Content: [Replacing Device Console (DevCon.exe) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/devcon-migration)
* Content Source: [windows-driver-docs-pr/devtest/devcon-migration.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/devcon-migration.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

forderud, 2023-09-21:
One thought: I'm only test-signing the driver when conducting the experiment above. Maybe the problem could be related to test-signing?

emilyhobby, 2023-09-21:
From the pnputil output, it appears as though the devices in question have a better matching driver installed, or otherwise do not meet the criteria for installation. The filter driver could rely on another driver being installed on the device before installation. To test this, create your test device, reboot the machine, then run the pnputil command to add and install the driver. If that isn't successful, run the following from an elevated command prompt to get a better idea of the test device, any matching drivers on the system, and their rank:

pnputil /enum-devices /instanceid <devgen device instance ID ROOT\DEVGEN\{...}> /drivers

I'll continue to track this issue and its resolution to inform any documentation updates.

forderud, 2023-09-22:
I'm still experiencing the same problem. Here's my output when running `pnputil /enum-devices` after trying to install the driver with `pnputil`:
```
> pnputil /enum-devices /instanceid "ROOT\DEVGEN\{0537181F-703E-5A4A-A90D-9E8F09E3D92E}" /drivers
Microsoft PnP Utility

Instance ID:                ROOT\DEVGEN\{0537181F-703E-5A4A-A90D-9E8F09E3D92E}
Device Description:         Microsoft USB IntelliMouse Explorer 3.0
Class Name:                 Mouse
Class GUID:                 {4d36e96f-e325-11ce-bfc1-08002be10318}
Manufacturer Name:          Microsoft
Status:                     Problem
Problem Code:               10 (0x0A) [CM_PROB_FAILED_START]
Problem Status:             0xC0000010
Driver Name:                msmouse.inf
Matching Drivers:
    Driver Name:            msmouse.inf
    Provider Name:          Microsoft
    Class Name              Mouse
    Class GUID:             {4d36e96f-e325-11ce-bfc1-08002be10318}
    Driver Version:         06/21/2006 10.0.19041.1
    Matching Device ID:     HID\Vid_045E&Pid_0047
    Driver Rank:            00FF1000
    Driver Status:          Best Ranked / Installed

    Driver Name:            oem44.inf
    Original Name:          firefly.inf
    Provider Name:          TODO-Set-Provider
    Class Name              Mouse
    Class GUID:             {4d36e96f-e325-11ce-bfc1-08002be10318}
    Driver Version:         09/22/2023 12.26.3.732
    Matching Device ID:     HID\Vid_045E&Pid_0047
    Driver Rank:            80FF0000
    Driver Status:          Outranked
```

Seems like the problem is caused by my locally built and test-signed [firefly](https://github.com/microsoft/Windows-driver-samples/tree/main/hid/firefly) filter driver is outranked by the in-built "msmouse" driver. This problem does _not_ occur when using `devcon install` for driver installation.

tedhudek, 2023-10-03:
Hi again Fredrik,

While it is subject to change, what you are seeing is the current expected behavior.

I added a Note describing the behavior to https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/pnputil-command-syntax#add-driver.  Thanks for helping us improve the docs!

closed 2023-10-03 by tedhudek

forderud, 2023-10-04:
@tedhudek Thanks for the clarification. Does this mean that I'm stuck with "devcon install" for the the [firefly](https://github.com/microsoft/Windows-driver-samples/tree/main/hid/firefly) sample for now, or is there a way to migrate to devgen/pnputil?

forderud, 2024-02-21:
Status update:
My problem was solved by using CertMgr.exe to install the driver certificate to the `root` and `trustedpublisher` stores as documented on https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/certmgr#example . Driver installation _without_ devcon afterwards worked like a charm.

Commands used:
```
CertMgr /add firefly.cer /s /r localMachine root 
CertMgr /add firefly.cer /s /r localMachine trustedpublisher
```