MicrosoftDocs/windows-driver-docs #3647: Build a basic v4 printer driver - Error parsing UMDF version string - ..0

Torxed, 2023-09-20:
The docs for getting started with basic v4 printer drivers lacks information, leading to:
```
1>Building 'MyV4PrintDriver' with toolset 'WindowsUserModeDriver10.0' and the 'Desktop' target platform.
1>Error parsing UMDF version string - ..0
>C:\Program Files (x86)\Windows Kits\10\build\10.0.22621.0\WindowsDriver.common.targets(559,5): error MSB6006: "stampinf.exe" exited with code 87.
1>Done building project "MyV4PrintDriver.vcxproj" -- FAILED.
```

More specifically it's the global variables `UMDF_VERSION_MAJOR` and `UMDF_VERSION_MINOR` that is not set:
[image]

Which I guess should be set with:
```c++
#define UMDF_VERSION_DATA Microsoft_WDF_UMDF_Version = { 2, 33, 0 }
```

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 227ceb38-396a-aa19-6304-be3df9b0ca08
* Version Independent ID: 2f34593f-aa20-384a-b1bb-c8a962ec4706
* Content: [Build a basic v4 printer driver - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/print/building-a-basic-v4-printer-driver)
* Content Source: [windows-driver-docs-pr/print/building-a-basic-v4-printer-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/print/building-a-basic-v4-printer-driver.md)
* Product: **windows-hardware**
* Technology: **print**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

chaomiosoft, 2023-11-13:
I modified the UMDF Version Number Property's value to "2.33.0". Because I use latest WDK. (Please refer below url https://learn.microsoft.com/en-us/windows-hardware/drivers/wdf/umdf-version-history)
So, I didn't get a build error message of UMDF.  But I got  a another error message of InfGate.exe.

Then, I modified path of command-Line of After build event in property page of project.
To full path of InfGate.exe. "$(WDKContentRoot)Tools\10.0.22621.0\x64\INFGate.exe" (if you select x64 driver env.)

I can build the sample of v4 printer driver.

barrygolden, 2024-01-12:
@Torxed, thanks for your feedback on this topic. As noted, this topic contains multiple errors and is not usable at this time.

Microsoft has announced end of life servicing for legacy v3 and v4 printer drivers. For more information, see [End of servicing plan for third-party printer drivers on Windows](https://learn.microsoft.com/windows-hardware/drivers/print/end-of-servicing-plan-for-third-party-printer-drivers-on-windows).

It is recommended to use Microsoft's IPP inbox class driver, along with Print Support Apps (PSA), to customize the print experience in Windows 10 and 11 for printer device development. For more information, see the [Print support app design guide](https://learn.microsoft.com/windows-hardware/drivers/devapps/print-support-app-design-guide).

closed 2024-01-12 by barrygolden