MicrosoftDocs/windows-driver-docs #3648: Missing WDK tools such as tracewpp.exe and INFGate.exe

Torxed, 2023-09-20:
As pointed out by others: 
* https://stackoverflow.com/questions/75247905/build-basic-v4-printer-driver-windows-error-infgate-exe-wdk

The page is missing a clear reference of how to resolve WDK issues.
For instance `C:\Program Files (x86)\Windows Kits\10\Tools\x86\` does not contain anything but a SecureBoot folder on Visual Studio 2022 following the steps in https://learn.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk which is a reference from https://learn.microsoft.com/en-us/windows-hardware/drivers/print/building-a-basic-v4-printer-driver. I assume this known as there's a note about:

> This printer driver tutorial contains multiple errors and is not usable at this time.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 227ceb38-396a-aa19-6304-be3df9b0ca08
* Version Independent ID: 2f34593f-aa20-384a-b1bb-c8a962ec4706
* Content: [Build a basic v4 printer driver - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/print/building-a-basic-v4-printer-driver)
* Content Source: [windows-driver-docs-pr/print/building-a-basic-v4-printer-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/print/building-a-basic-v4-printer-driver.md)
* Product: **windows-hardware**
* Technology: **print**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

RAJU2529, 2023-09-20:
@Torxed , 
location of  **infgate.exe**
C:\Program Files (x86)\Windows Kits\10\Tools\10.0.22621.0\x64

location  of  **tracewpp.exe** 
C:\Program Files (x86)\Windows Kits\10\bin\10.0.22621.0\x64

You need to install the latest version of WDK 22621  for windows 11.

Torxed, 2023-09-20:
> C:\Program Files (x86)\Windows Kits\10\Tools\10.0.22621.0\x64

I appreciate the information, but this is not the location that Visual Studio is looking for the binary. So either the docs should mention how to change the location accurately - or Visual Studio needs to adapt and find the binary in the correct location.

The [guide](https://learn.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk#download-and-install-the-windows-11-version-22h2-wdk) has a fix for the location issue that does not work:

> C:\Program Files (x86)\Windows Kits\10\Vsix\VS2022\10.0.22621.382\WDK.vsix

All you'll get is:

[image]

And even if you manually try to install the `inf`, the `infgate.exe` is missing the `/WDK` option:

[image]

So I'm not even sure how Visual Studio would be able to call the binary even if it found it?

barrygolden, 2024-01-12:
@Torxed, thanks for your feedback on this topic. As noted, this topic contains multiple errors and is not usable at this time.

Microsoft has announced end of life servicing for legacy v3 and v4 printer drivers. For more information, see [End of servicing plan for third-party printer drivers on Windows](https://learn.microsoft.com/windows-hardware/drivers/print/end-of-servicing-plan-for-third-party-printer-drivers-on-windows).

It is recommended to use Microsoft's IPP inbox class driver, along with Print Support Apps (PSA), to customize the print experience in Windows 10 and 11 for printer device development. For more information, see the [Print support app design guide](https://learn.microsoft.com/windows-hardware/drivers/devapps/print-support-app-design-guide).

closed 2024-01-12 by barrygolden