MicrosoftDocs/windows-driver-docs #3651: acquireConnectionAsync error, as connectionProifle grows

hunter1303, 2023-09-22:
[Enter feedback here]
If you use acquireConnectionAsync to establish a connection and only set the authentication method, The parameter is incorrect errors will occur randomly.
In addition, I found that async will cause the system's connectionProfile to increase cumulatively. Is this a bug, or is it being used incorrectly?

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 2bb57675-1890-5529-2e52-be19aeabe221
* Version Independent ID: caee0f97-a9b4-fdf5-c3c2-f3474786e4d0
* Content: [Developing apps using multiple PDP contexts - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/mobilebroadband/developing-apps-using-multiple-pdp-contexts)
* Content Source: [windows-driver-docs-pr/mobilebroadband/developing-apps-using-multiple-pdp-contexts.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/mobilebroadband/developing-apps-using-multiple-pdp-contexts.md)
* Product: **windows-hardware**
* Technology: **mobilebroadband**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

hunter1303, 2023-09-23:
in face, I tried to use multiple pdp to develop a demo based on 5G slice. Multiple pdp should be the basic function. On this basis, URSP and other functions are added. Currently, the mbb api still has missing functions here.

hunter1303, 2023-09-23:
pseudocode：
CellularApnContext cellularApnContext;
cellularApnContext.AccessPointName("AccessPointName");	//AccessPointName base operator
cellularApnContext.AuthenticationType(CellularApnAuthenticationType::None);
auto op{ ConnectivityManager::AcquireConnectionAsync(cellularApnContext) };
op.get();
Sometimes it's successful, sometimes it's throw exception with "The parameter is incorrect".
I really don't know how to diagnose the error here。

aviviano, 2023-12-15:
Hi @hunter1303, 
Thanks for opening this issue. I sent your feedback to the development team, and they said: "The Cellular Context just gets passed through WCM to WWAN."
Please let me know if you are still having issues. 
Thanks.

aviviano, 2023-12-22:
Closing due to inactivity. Please open a new issue if you have further documentation issues or questions. 
Thank you.

closed 2023-12-22 by aviviano