MicrosoftDocs/windows-driver-docs #3655: incorrect information

PeterSommerlad, 2023-09-23:
"There are no division operations because there are no overflow conditions."

for division there are one or two cases causing erroneous results:

1. division by zero

2. for signed integers: division of the smallest value (INTMIN etc) by -1 which causes overflow 

so the above text is wrong stating there are no overflow confitions

hsebs, 2023-09-23:
Are you talking about [Using Safe Integer Functions](https://learn.microsoft.com/en-us/windows-hardware/drivers/kernel/ntintsafe-design-guide)?

These functions are provided to prevent operation overflows, not buffer overflows.

In first case, Division by zero raises arithmetic exception, not arithmetic overflow.
In second case, Division by -1 does not cause arithmetic overflow.

PeterSommerlad, 2023-09-23:
I doubt that statement

in two's complement the most negative number cannot be multiplied or divided by minus one, because it cannot be represented with negated (now positive) sign. I would call that overflow. with wrapping the resulting value is mathematically incorrect eg -2^31 instead of 2^31 on a 32bit integer.

hsebs, 2023-09-24:
You are right. I was wrong about the second case.

barrygolden, 2023-09-28:
The incorrect text ("There are no division operations because there are no overflow conditions.") has been removed.

closed 2023-09-28 by barrygolden