MicrosoftDocs/windows-driver-docs #3656: PcRegisterSubdevice failed with STATUS_ALLOTTED_SPACE_EXCEEDED

SlimSnow, 2023-09-25:


tedhudek, 2023-09-25:
Sorry, can you provide more info about what you are requesting?

SlimSnow, 2023-09-26:
Sorry,I shoud provide more info previously.
The situation is that I am a newer to windows kernel driver and I want to dynamic add subdevices  a pair of mic and speaker.I have a app to send custom IOCTL code to the driver ,and when driver receive the code will add a pair of subdevices.But when I want to add the sixth pair of subdevices the API PcRegisterSubdevice failed with STATUS_ALLOTTED_SPACE_EXCEEDED.I think i pass valid parameters to PcRegisterSubdevice.
[image]
[image]
[image]
I am not sure is this issue caused by the subdevices name is the same,I wanna change the subdevices name, but like I said I am a newer to Windows kernel I do not faind a way to make subdevices name unique. Do you have any ideas?Thank you so muck and looking forward to your reply.

tedhudek, 2023-10-03:
Hi @SlimSnow,

There could be a limit of 5.  Or you could try unique names. But I do not have specific knowledge in this area, so I am not sure how to do that.  I would recommend posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive. This repo is for documentation, so if you discover anything that you think we should add to the docs, feel free to let us know.

* [Microsoft Q&A](https://learn.microsoft.com/en-us/answers/questions)
* [OSR's NTDEV forum](https://community.osr.com/categories/ntdev)

closed 2023-10-03 by tedhudek