MicrosoftDocs/windows-driver-docs #3669: 2nd Try - WinDbg Preview Please document how to set environment variables for Start Debugging > Launch executable

buzz3791, 2023-10-02:
Please document how to set environment variables for WinDbg Preview version 1.2306.12001.0 Start Debugging > Launch executable
I'm not asking about symbol .pdb files, I'm asking about supporting values retrieve via std::getenv().

I'm asking for documentation as follows:
Lets say I have a C++ app myapp.exe that contains code like:
```
#include
#include
int main()
{
  if(const char* env_var1 = std::getenv("MYVAR1")) {
    std::cout << "Your MYVAR1 is: " << env_var1 << '\n';
  }
}
```
When launching from a cmd.exe prompt I would do the following:
```
set MYVAR1 = "myvalue1"
myapp.exe
```
Then at runtime MYVAR1 would be myvalue1 so myapp.exe would output:
`Your MYVAR1 is: myvalue1`

I'm asking for documentation on how to configure WinDbg-Preview to launch myapp.exe so that WinDbg configures the environment for the myapp.exe process such that the `MYVAR1` environment variable exists and has the configured value `myvalue1`; just like the cmd.exe example.

How can I achieve this when using WinDBG-Preview to launch myapp.exe?

Note the Visual Studio documentation explains the feature I want for WinDbg-Preview here [1].

1 - "Environment row" in the table titled "Configuration Properties folder (Debugging category)" - https://learn.microsoft.com/en-us/visualstudio/debugger/project-settings-for-a-cpp-debug-configuration?view=vs-2022

tedhudek, 2023-10-02:
Originally surfaced in #3603

DOMARS, 2023-10-23:
@buzz3791 - Thank you for clarifying your interest in the use of environment variables to pass to an executable that will be debugged - I would assume in user mode with WinDbg. 

I will consider adding more information on the interplay between environmental variables and launching a debugger session, in the future, but all requests are balanced against our backlog of existing requests and prioritized for the widest customer impact.  We are not able to provide support for all possible use cases of WinDbg, the Windows OS and various programming languages, as valid as they may be.

In case it is helpful, this topic describes how to pass arguments to a launching executable when starting a user mode application via WinDbg - 
[WinDbg - Start a user mode session](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-user-mode-preview). Not sure if that technique can be used to pass your MYVAR1 as an argument to the app. 

You can use the [.shell](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/-shell--command-shell-) command to pass commands to a process being launched using .shell with the -ci option.

The example in the doc shows how to pass !process command output to a perl script: `.shell -ci "!process 0 7" perl.exe parsemyoutput.pl`

Another possible strategy might be to attach to a script file that sets the environment value, and then loads the target application. The tricky bit is that the context of the script may be on a different thread, then your app, so you may need to locate and move to that context, if that is possible. Some of this is described in topics such as [Threads and Processes]( https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/threads-and-processes).

As I mentioned previously, for specific use case scenarios of WinDbg, the forums, such as those provided by OSR are your best bet.
•	[OSR's windbg List](https://community.osr.com/categories/windbg)

I recommend both the Windows Internals books as well as *Inside Windows Debugging: Practical Debugging and Tracing Strategies* by Tarik Soulami. These resources as well as OSR, provide additional tips and techniques not covered in the documentation.

closed 2023-10-23 by DOMARS