MicrosoftDocs/windows-driver-docs #3674: no information about IRP_MJ_FILE_SYSTEM_CONTROL

karolKiedrowski, 2023-10-11:
There is no information about IRP_MJ_FILE_SYSTEM_CONTROL in this neither previos section.
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 39d4c36b-872d-6454-0525-46876caaa912
* Version Independent ID: 381982c3-c5a9-4f38-2a42-3c935fd506fa
* Content: [Optional Dispatch Routines - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/kernel/optional-dispatch-routines)
* Content Source: [windows-driver-docs-pr/kernel/optional-dispatch-routines.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/optional-dispatch-routines.md)
* Product: **windows-hardware**
* Technology: **kernel**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2023-10-11:
Thanks for the feedback @karolKiedrowski.  Can you provide more detail on what specifically you are trying to accomplish?  I think that will enable me to help you better.

karolKiedrowski, 2023-10-12:
I need an explanation about dispatch routine that is responsible for
processing of given irp message

tedhudek, 2023-10-12:
What type of driver are you writing?  Is it by any chance a file system driver or a file system filter driver?

You could try asking https://www.bing.com/chat for sample code for a dispatch routine.  I just tried it, and it gave me starter code based on the documentation and sample code it found.

tedhudek, 2023-10-23:
Closing out for now, please reopen if still needed.

closed 2023-10-23 by tedhudek