MicrosoftDocs/windows-driver-docs #3693: Windows Server 2022 CodeQL procedure does not work

sschmitz41, 2023-10-25:
The recommended codeql version of 2.4.6 with WHCP_21H2 branch for Windows Server 2022 does not work.  Windows 11 22H2 does work, but a many of the example command lines have obvious errors in them.  This page needs to be completely reviewed and verified.

For Windows Server 2022, the following error is encountered:

Warning: d:\codeql\Windows-Driver-Developer-Supplemental-Tools\codeql\codeql-queries\cpp\ql\src\qlpack.yml: Error reading d:\codeql\Windows-Driver-Developer-Supplemental-Tools\codeql\codeql-queries\cpp\ql\src\qlpack.yml: Unrecognized field "defaultSuiteFile" (class com.semmle.frontend.packs.QlPack), not marked as ignorable (14 known properties: "license", "dbscheme", "extractor", "libraryPathDependencies", "repository", "name", "upgrades", "version", "library", "description", "downgrades", "tests", "dependencies", "suites"])
 at [Source: (StringReader); line: 8, column: 54] (through reference chain: com.semmle.frontend.packs.QlPack["defaultSuiteFile"])
Warning: d:\codeql\Windows-Driver-Developer-Supplemental-Tools\codeql\codeql-queries\python\ql\src\qlpack.yml: Error reading d:\codeql\Windows-Driver-Developer-Supplemental-Tools\codeql\codeql-queries\python\ql\src\qlpack.yml: Unrecognized field "defaultSuiteFile" (class com.semmle.frontend.packs.QlPack), not marked as ignorable (14 known properties: "license", "dbscheme", "extractor", "libraryPathDependencies", "repository", "name", "upgrades", "version", "library", "description", "downgrades", "tests", "dependencies", "suites"])
 at [Source: (StringReader); line: 8, column: 57] (through reference chain: com.semmle.frontend.packs.QlPack["defaultSuiteFile"])
A fatal error occurred: The QL pack 'codeql-cpp' which is referenced from d:\codeql\Windows-Driver-Developer-Supplemental-Tools\codeql\windows-drivers\suites\windows_driver_mustfix.qls cannot be found.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 5660df59-18ee-441d-afc7-023da97a7da5
* Version Independent ID: 1e1c4d4d-7c4b-6af2-64f2-d51162a8992f
* Content: [CodeQL and the Static Tools Logo Test - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/static-tools-and-codeql)
* Content Source: [windows-driver-docs-pr/devtest/static-tools-and-codeql.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/static-tools-and-codeql.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2023-11-20:
@sschmitz41 - Thanks for this PR indicating that updates are possibly needed for CodeQL. I'm checking to see if we can update to 
2.15.1 version for the latest WDK. 

The https://github.com/microsoft/Windows-Driver-Developer-Supplemental-Tools page, indicates to use 2.11.5 for general use.

DOMARS, 2024-03-01:
@sschmitz41 - Thanks again for this PR. I have an internal tracking item to update the docs page, and hope to have that work done shortly, so will close this issue. In the meantime, please refer to the updated directions here: https://github.com/microsoft/Windows-Driver-Developer-Supplemental-Tools
There are now directions there for Server 2022 that point you towards the correct CLI version, qlpack and branch to use. The docs will reflect this information, but that release matrix should be the key piece of information that you need for Server testing. Thanks again for filing this issue, and using CodeQL to improve your code.

closed 2024-03-01 by DOMARS